letscontrolit/ESPEasy #2735: Rule Question

fly74, 2019-11-12:
I would like to have a rule like this:

`
On System#Boot do
    GPIO,0,0
    GPIO,2,1
endon

On pon do
 gpio,0,1
 gpio,2,0
 gpio,15,1
endon

On poff do
 gpio,0,0
 gpio,2,1
 gpio,15,0
endon

On SW#State do
if [REL#State] = 0 then
    pon
else
    poff
endif
enddo
`

But the event start from the switch don't work, what's wrong? Can event called from switch event?

TD-er, 2019-11-12:
You should use the command `event` to call an event trigger.
See the [Command Reference](https://espeasy.readthedocs.io/en/latest/Reference/Command.html)

fly74, 2019-11-12:
ah that works, thx:

...
On SW#State do
if [REL#State] = 0 then
    Event,pon
else
    Event,poff
endif
enddo

closed 2019-11-12 by fly74