letscontrolit/ESPEasy #2736: Release mega-20191113

giobbe70, 2019-11-13:
Sensor DS18b20 and DH11/DH22 not work
Thanks

Giobbe

### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

giobbe70, 2019-11-13:
Release mega-20191113

DS18b20 and DH11/DH22 not work

uzi18, 2019-11-13:
please give us more info like connections you use, task configuration and debug log

TD-er, 2019-11-13:
And what build did work with your setup?
So in other words, exactly what we are asking for in the issue template.

TD-er, 2019-11-16:
With this amount of information we cannot help, so I will close this issue.
Please provide the absolute minimum of information, like what version did work, how is it connected, etc.
Also the topic title is absolutely not describing what the issue is about.

closed 2019-11-16 by TD-er