letscontrolit/ESPEasy #2741: Rules bug? Or need to rewrite?

Budman1758, 2019-11-14:
This is pretty simple I hope.  When using the latest release ( 20191113 ) This rule command does not work...

> SendToHttp 192.168.1.61,80,/control?cmd=Event,Close
Produces this in the log...

> Command: sendtohttp
36287204: Too many arguments: cmd=SendToHttp arg1=192.168.1.61 arg2=80 arg3=/control?cmd=event extraArg4=close lineLength=51
36287206: Line: _SendToHttp 192.168.1.61,80,/control?cmd=Event,Close

When using release ( 20191109 ) the command works fine and sends the event to the target unit as its supposed to.

ESP_Easy_mega-20191108_test_core_260_sdk3_alpha_ESP8266_4M1M.bin  < using this bin file from both releases.  I need test version because this is a keypad unit using HT16K33 chip. 

I know there has been a lot of rules parsing work lately so dunno if it's a bug or a new "feature"  :)

uzi18, 2019-11-14:
try to change it to
`SendToHttp 192.168.1.61,80,"/control?cmd=Event,Close"`

TD-er, 2019-11-14:
The main change is that we no longer parse the commands like "well use whatever you may find on the line", but just look for a strict parameter position.
And the comma you use in your URL also happens to be a parameter separator.
So by using quotes you force the comma to be part of the parameter.

TD-er, 2019-11-14:
Can you close the issue if it appears to be working with @uzi18 's suggestion?

Budman1758, 2019-11-14:
> try to change it to
> `SendToHttp 192.168.1.61,80,"/control?cmd=Event,Close"`
> 
> we strict check now how many arguments are passed to command

That does the trick.

closed 2019-11-14 by Budman1758