letscontrolit/ESPEasy #2742: Nomal Core 241 unneeded !! (for pwm)

DittelHome, 2019-11-15:
Today i tested again pwm with core 2.5.2. Suprise, pwm is working **without** flackering.
I tried this version:
ESP_Easy_mega-20191113_normal_ESP8266_4M1M.bin

This means, we could/should delete the ESP_Easy_mega-20191113_normal_core_241_ESP8266_4M1M.bin.

TD-er, 2019-11-15:
That's good to hear.
Can you also test it with the core 2.6.0?
Just to be sure we also don't need core 2.5.2 just for PWM.

Now the only reason to keep 2.4.x is the size of the minimal OTA versions, since later core libs only become bigger and bigger. :(

DittelHome, 2019-11-15:
ESP_Easy_mega-20191113_normal_core_260_sdk222_alpha_ESP8266_4M1M.bin is also working fine **without** flickering !!

TD-er, 2019-11-15:
Great!
So we can get rid of those core builds too, when core 2.6.0 can be built directly from PIO without some tricks.

Misiu, 2019-11-15:
@TD-er [this ](https://github.com/platformio/platform-espressif8266/issues/167)is the related issue for PIO?
Also, core 2.6.1 is out :)

TD-er, 2019-11-15:
> @TD-er [this ](https://github.com/platformio/platform-espressif8266/issues/167)is the related issue for PIO?
> Also, core 2.6.1 is out :)

Yep, that's the issue.
And when I tried building 2.6.1 yesterday I had a build issue, will try again.

TD-er, 2019-11-15:
@Misiu See: https://github.com/esp8266/Arduino/issues/6779

PR #2744 (2019-11-21): [PIO] Cleanup platformio.ini and move to core 2.6.1 as default (#2742)
Fixes: #2742
commit 0b499cce (2019-11-21): [PIO] Cleanup platformio.ini and move to core 2.6.1 as default (#2742)

Fixes: #2742
closed 2019-11-21 by TD-er