letscontrolit/ESPEasy #2745: Sensor DS18b20 and DH11/DH22 not work

giobbe70, 2019-11-16:
On Release mega-20191116 Sensor DS18b20 and DH11/DH22 not work,
Its value NaN.
[image]
[image]
[image]

giobbe70, 2019-11-16:
[image]

TD-er, 2019-11-16:
And in what ESPEasy version did it work well?

giobbe70, 2019-11-16:
All version on Release 20191116
I have install 

  | ESP_Easy_mega-20191116_normal_core_241_ESP8266_4M1M.bin

giobbe70, 2019-11-16:
the 20190630 version work well

TD-er, 2019-11-16:
Have you used older versions of ESPEasy where it was running well?

For example this is on my breadboard, to which I just flashed the latest code:
[image]

giobbe70, 2019-11-16:
ESP_Easy_mega-20191003_test_core_260_sdk3_alpha_ESP8266_4M1M.bin
WORK WELL

TD-er, 2019-11-16:
The topic title suggests the Dallas sensors also don't work, but like I showed you, they work on my setup.
So if it then both the Dallas sensors and the DHT11/DHT22? Or only the last one?

giobbe70, 2019-11-16:
Inglese

IN MY CONFIGURATION WITH THE 20190630 VERSION THE Dallas sensors and the DHT11 / DHT22 WORK WELL. IN NEW VERSIONS FROM SEPTEMBER TO FORWARD DO NOT WORK. THE VALUE IS NaN

uzi18, 2019-11-16:
please provide debug log

giobbe70, 2019-11-16:
64974: EVENT: Clock#Time=Sat,13:43
65860: SW   : State 0.00
65863: EVENT: RELE-TERM-SOTTO#Switch=0.00
67861: SW   : State 0.00
67862: EVENT: RELE-TERM-SOTTO#Switch=0.00
68171: EVENT: TEMP-PANNELLO#Temperature=38.50
68749: EVENT: TEMP-BOILER#Temperature=78.00
69862: SW   : State 0.00
69864: EVENT: RELE-TERM-SOTTO#Switch=0.00
71860: SW   : State 0.00
71862: EVENT: RELE-TERM-SOTTO#Switch=0.00
73242: EVENT: TEMP-PANNELLO#Temperature=38.50
73872: SW   : State 0.00
73873: EVENT: RELE-TERM-SOTTO#Switch=0.00
74270: EVENT: TEMP-BOILER#Temperature=78.00
75720: DHT  : No Reading
75860: SW   : State 0.00
75862: EVENT: RELE-TERM-SOTTO#Switch=0.00
77715: WD   : Uptime 1 ConnectFailures 0 FreeMem 16456 WiFiStatus 3
77876: SW   : State 0.00
77878: EVENT: RELE-TERM-SOTTO#Switch=0.00
78394: EVENT: TEMP-PANNELLO#Temperature=38.50
79860: SW   : State 0.00
79862: EVENT: RELE-TERM-SOTTO#Switch=0.00
80096: EVENT: TEMP-BOILER#Temperature=78.00
81860: SW   : State 0.00
81862: EVENT: RELE-TERM-SOTTO#Switch=0.00
83474: EVENT: TEMP-PANNELLO#Temperature=38.50
83897: SW   : State 0.00
83899: EVENT: RELE-TERM-SOTTO#Switch=0.00
85860: SW   : State 0.00
85862: EVENT: RELE-TERM-SOTTO#Switch=0.00
86077: EVENT: TEMP-BOILER#Temperature=78.00
87860: SW   : State 0.00
87862: EVENT: RELE-TERM-SOTTO#Switch=0.00
88748: EVENT: TEMP-PANNELLO#Temperature=38.50
89860: SW   : State 0.00
89862: EVENT: RELE-TERM-SOTTO#Switch=0.00
91789: EVENT: TEMP-BOILER#Temperature=78.00
92006: SW   : State 0.00
92008: EVENT: RELE-TERM-SOTTO#Switch=0.00
93713: EVENT: TEMP-PANNELLO#Temperature=38.50
93931: SW   : State 0.00
93933: EVENT: RELE-TERM-SOTTO#Switch=0.00
95860: SW   : State 0.00
95862: EVENT: RELE-TERM-SOTTO#Switch=0.00
97705: EVENT: TEMP-BOILER#Temperature=78.00
97947: SW   : State 0.00
97952: EVENT: RELE-TERM-SOTTO#Switch=0.00
98818: EVENT: TEMP-PANNELLO#Temperature=38.50
99860: SW   : State 0.00
99862: EVENT: RELE-TERM-SOTTO#Switch=0.00
101860: SW   : State 0.00
101862: EVENT: RELE-TERM-SOTTO#Switch=0.00
103762: EVENT: TEMP-BOILER#Temperature=78.00
103970: EVENT: TEMP-PANNELLO#Temperature=38.50
104185: SW   : State 0.00
104187: EVENT: RELE-TERM-SOTTO#Switch=0.00
105738: DHT  : No Reading
105861: SW   : State 0.00
105863: EVENT: RELE-TERM-SOTTO#Switch=0.00
107715: WD   : Uptime 2 ConnectFailures 0 FreeMem 16424 WiFiStatus 3
107860: SW   : State 0.00
107862: EVENT: RELE-TERM-SOTTO#Switch=0.00
109045: EVENT: TEMP-PANNELLO#Temperature=38.50
109487: EVENT: TEMP-BOILER#Temperature=78.00
109917: SW   : State 0.00
109919: EVENT: RELE-TERM-SOTTO#Switch=0.00
111860: SW   : State 0.00
111862: EVENT: RELE-TERM-SOTTO#Switch=0.00

giobbe70, 2019-11-16:
BOILER SENSOR IS REMOTE SENSOR ON ANOTHER ESPEASY WITH EASPEASY 20190630

TD-er, 2019-11-16:
You need to get logs from the unit that actually has the sensor connected.
From remote sensors it is next to impossible to see what is wrong, since you only see the reported value, not why a value is incorrect.

giobbe70, 2019-11-16:
the local sensor DH22 not work .
The sensor DS18b20 is remote on other ESPEASY release 20190630.
If I load the Release 201911013 The sensor DS18b20 not work.
the log is telease 20191116 with DH22 not work

TD-er, 2019-11-16:
Well the plugin P005_DHT was last changed in August, so I guess it is not something in the plugin itself.

Does the plugin give data sometimes? Or are all measurements failing?

giobbe70, 2019-11-16:
DH22 and  DS18b20 all measurements failing after August.
Thanks

TD-er, 2019-11-16:
Which August build?
There have been several changes on the DHT plugin in August.
See: https://github.com/letscontrolit/ESPEasy/commits/mega/src/_P005_DHT.ino

giobbe70, 2019-11-16:
I don't know that in October I am in agreement with the sensor malfunction

uzi18, 2019-11-16:
DHT22 works for me with latest code 
@giobbe70 please try to change sensor type, and give us info what readings you have on other settings

@TD-er it is first report about not working sensor from long time, in my opinion it is hardware related
we can add some more debug for "no reading" to have more info for the future, maybe we can move noInterrupts() 2 lines up.
 
https://github.com/letscontrolit/ESPEasy/blob/0d6d1366996335ec20187c8ba4f4e4db747fac18/src/_P005_DHT.ino#L175-L179

giobbe70, 2019-11-16:
Hi,
Unfortunately I cannot change settings, as the sensor is in a closed thermostat

TD-er, 2019-11-16:
@uzi18 Moving the interrupts up is fine, as long as you make sure they are enabled before exiting the function.
Still, I wonder why it was working fine before.

Also, what is the common change for both Dallas and DHT plugins?
They can send out NaN, maybe the handling of that is the problem?

clumsy-stefan, 2019-11-16:
@TD-er not sure if it's related, but roughly after 20191108 I experience also again more strange issues, like laggy-webpages, HW-WD reboots, Units that won't connect to AP anymore (especially when signal is weak), NeoPixel's don't run smoothly anymore, etc.

To me it looks like timing issues or background tasts that do not get executed or stalled... Which could explain why Dallas sensors are not read anymore correctly (wrong timing etc.)...

Can't really nail it currently, I'm searching, but did something change in WiFi handling or handling of background tasks?

uzi18, 2019-11-16:
@TD-er small tweaks here #2748

clumsy-stefan, 2019-11-17:
> maybe it is core related?

Could very well be, yes, as I'm using latest GIT core... I'm still trying to track down what exactly happens... It could also be memory related, as the 1M units have more problems than the 4/16M ones...

TD-er, 2019-11-17:
1M units often have more low-cost flash chips. Maybe we should lower the flash frequency on those? Or maybe it is lowered and that's what does make it more slow.

clumsy-stefan, 2019-11-17:
strange thing is, it happens only since about a week ago or so (after 20191108) and I have a 16M unit with some Dallas sensors and 50 Neopixels. I see wrong colors on the strip or laggy movement of the color pattern, which normally means some timing is not correct when driving the neo's....

TD-er, 2019-11-17:
This 16M module, what SPIFFS size do you use?
Maybe this is not related to this issue, since the reported issues here apparently started in August (and nobody else has them?)

clumsy-stefan, 2019-11-17:
14M SPIFFS.

Strangely part of the above described phenomens vanished... All untis stay connected, just one or so HW watchdog... Only the Neopixel's still seem to have problems... but all traces so far show no anormality... so let's say this is a very local problem to me (at least until I further evidence of something wrong going on...

TD-er, 2019-11-17:
> 14M SPIFFS.

That's slow as h*ll.

You should not use SPIFFS for sizes > 2 MB.
It is also noted in the platform.ini file.

clumsy-stefan, 2019-11-17:
Hmm... thanks for pointing... Never had an issue until now... I'll try with a 2MB SPIFFS or what would you suggest?

Anyway, that's just one out of ~40 running units... normally I use 4M with 1M SPIFFS.

TD-er, 2019-11-17:
Smaller SPIFFS is more speed.
SPIFFS is terrible inefficient.

One of the optimizations/fixes I made recently is to make sure the extra task settings are loaded when needed (maybe still a bit too often), but at least it is loaded when needed and that wasn't always correct.

clumsy-stefan, 2019-11-18:
Arduino only officially supports 14M and 15M FS sizes:
[image]

So I'd need to generate a complete new config if I want to use other FS size....

TD-er, 2019-11-18:
Or use 4 MB size?

clumsy-stefan, 2019-11-18:
> Or use 4 MB size?

could try that, yes...

giobbe70, 2019-11-20:
On Release 20191119 another NOT WORK DH 22 and Dallas sensors

uzi18, 2019-11-20:
please set in advenced options "weblog" to "debug" and paste here log

giobbe70, 2019-11-20:
> 
> 
> @giobbe70 please set in advanced options "weblog" to "debug" and paste here log

92854: EVENT: TEMP-BOILER#Temperature=71.50
93823: SW   : State 0.00
93824: EVENT: RELE-TERM-SOTTO#Switch=0.00
95737: EVENT: TEMP-PANNELLO#Temperature=55.00
95841: SW   : State 0.00
95843: EVENT: RELE-TERM-SOTTO#Switch=0.00
97822: SW   : State 0.00
97823: EVENT: RELE-TERM-SOTTO#Switch=0.00
98656: EVENT: TEMP-BOILER#Temperature=71.56
99822: SW   : State 0.00
99823: EVENT: RELE-TERM-SOTTO#Switch=0.00
100852: EVENT: TEMP-PANNELLO#Temperature=55.00
101822: SW   : State 0.00
101823: EVENT: RELE-TERM-SOTTO#Switch=0.00
103644: DHT  : No Reading
103822: SW   : State 0.00
103824: EVENT: RELE-TERM-SOTTO#Switch=0.00
104941: EVENT: TEMP-BOILER#Temperature=71.56
105161: WD   : Uptime 2 ConnectFailures 0 FreeMem 18496 WiFiStatus 3
105822: SW   : State 0.00
105823: EVENT: RELE-TERM-SOTTO#Switch=0.00
105983: EVENT: TEMP-PANNELLO#Temperature=55.00
106935: EVENT: Clock#Time=Wed,21:29
107822: SW   : State 0.00
107823: EVENT: RELE-TERM-SOTTO#Switch=0.00

aleksey-ru52, 2019-11-21:
I'm noticed strange behaviour of DS18b20 with fw mega-20191116 too.
One probe - works just fine. Two or more probes - readings changed very slow and unpredictable, not as you set in "interval". First time i thought it's a resolution problem - readings changed very slow and for a big step. Then i suppose probe is somehow broken. I changed probe, but it did not helped. Then i suppose it's a power problem or something. Now i'm sure it deals with how many probes are enabled. I'm just disable 4 probes and last one work just fine. Hardware setup does not changes. All 5 sensors are close to each other at breadboard. ESP board is 4M esp8266 devboard.
mega-20191003 did not have this issue. Want to try latest one at that moment.

uzi18, 2019-11-21:
@aleksey-ru52 is it possible for you to paste here debug log, so we can check it ?

TD-er, 2019-11-21:
One of the symptoms mentioned online about fake Dallas sensors is that it does not work well with multiple sensors (>3 is mentioned)

Is there a possibility these sensors are forgeries?

uzi18, 2019-11-21:
@giobbe70 you need to change log level for weblog from info to debug please

aleksey-ru52, 2019-11-22:
> One of the symptoms mentioned online about fake Dallas sensors is that it does not work well with multiple sensors (>3 is mentioned)
> 
> Is there a possibility these sensors are forgeries?
well, it's possible... but they all work as they suppose to work with mentioned mega-20191003

TD-er, 2019-11-22:
> well, it's possible... but they all work as they suppose to work with mentioned mega-20191003

Yep that's still puzzling me.
It is really hard to tell what changes resulting in issues are due to fake sensors and which ones are our own bugs.

giobbe70, 2019-11-22:
139742: SW   : State 0.00
139743: EVENT: RELE-TERM-SOTTO#Switch=0.00
141220: EVENT: TEMP-BOILER#Temperature=77.56
141720: SW   : State 0.00
141721: EVENT: RELE-TERM-SOTTO#Switch=0.00
143721: SW   : State 0.00
143722: EVENT: RELE-TERM-SOTTO#Switch=0.00
144560: EVENT: TEMP-PANNELLO#Temperature=61.50
145721: SW   : State 0.00
145722: EVENT: RELE-TERM-SOTTO#Switch=0.00
147069: EVENT: TEMP-BOILER#Temperature=77.56
147720: SW   : State 0.00
147722: EVENT: RELE-TERM-SOTTO#Switch=0.00
149607: EVENT: TEMP-PANNELLO#Temperature=61.00
149738: SW   : State 0.00
149739: EVENT: RELE-TERM-SOTTO#Switch=0.00
151721: SW   : State 0.00
151722: EVENT: RELE-TERM-SOTTO#Switch=0.00
153008: EVENT: TEMP-BOILER#Temperature=77.56
153721: SW   : State 0.00
153722: EVENT: RELE-TERM-SOTTO#Switch=0.00
154722: EVENT: TEMP-PANNELLO#Temperature=61.00
155720: SW   : State 0.00
155722: EVENT: RELE-TERM-SOTTO#Switch=0.00
157720: SW   : State 0.00
157721: EVENT: RELE-TERM-SOTTO#Switch=0.00
158858: EVENT: TEMP-BOILER#Temperature=77.56
159720: SW   : State 0.00
159722: EVENT: RELE-TERM-SOTTO#Switch=0.00
159855: EVENT: TEMP-PANNELLO#Temperature=61.50
161720: SW   : State 0.00
161722: EVENT: RELE-TERM-SOTTO#Switch=0.00
163720: SW   : State 0.00
163721: EVENT: RELE-TERM-SOTTO#Switch=0.00
164687: EVENT: TEMP-BOILER#Temperature=77.56
164969: EVENT: TEMP-PANNELLO#Temperature=61.50
165566: DHT  : No Reading
165721: SW   : State 0.00
165722: EVENT: RELE-TERM-SOTTO#Switch=0.00
167228: WD   : Uptime 3 ConnectFailures 0 FreeMem 15408 WiFiStatus 3
167720: SW   : State 0.00
167722: EVENT: RELE-TERM-SOTTO#Switch=0.00
168866: EVENT: Clock#Time=Fri,13:33
169720: SW   : State 0.00
169721: EVENT: RELE-TERM-SOTTO#Switch=0.00
170083: EVENT: TEMP-PANNELLO#Temperature=61.00
170559: EVENT: TEMP-BOILER#Temperature=77.56
171720: SW   : State 0.00
171722: EVENT: RELE-TERM-SOTTO#Switch=0.00
173722: SW   : State 0.00
173723: EVENT: RELE-TERM-SOTTO#Switch=0.00

aleksey-ru52, 2019-11-24:
Well, i should admit, in my case problem is hw setup, not firmware. Even chineese probes works good. I have two similar setups - one for testing and one for "prod" - it's connected to my house central heating. These setups have similar settings and probes count. Then last time i updated "testing" setup  at the same time i've changed interrupt pin for one-wire, this new pin have led and i can easily "see" then bus is polling probes. Well, led or pull-up resistor for this led or something else - is the problem, or i should lover one wire parasite power resistor. Yesterday i connect one wire to another pin, without led and i did not noticed any difference in fw mega-20191003 and mega-20191119. Even "bad" probes working just fine with high poll frequency. They shows different values - not a big deal, but they work all together.
Sorry again, you a right.

uzi18, 2019-11-24:
so try lower resistor maybe it will work with led after this change.

TD-er, 2019-11-24:
I advice to use a transistor to switch the LED if you want to monitor the pin activity.
The problem is probably related to the voltage drop over the LED, which makes the presence of the LED kind of difficult to predict.

jeroenhendricksen, 2019-12-05:
I experienced the same problem as OP with my DHT22 sensor, going from firmware version file `ESP_Easy_mega-20190425_normal_ESP8266_4M.bin` to `ESP_Easy_mega-20191130_normal_ESP8266_4M1M.bin` on a `NodeMCU v3 CH340G`.
After the firmware upgrade, the humidity and temperature sensor both gave me `NaN`. Then I figured I switch the GPIO port from GPIO-16 (D0) to GPIO-15 (D8) which fixed my problem. I'm mentioning this here because it might help someone fix their sensor problem, just by avoiding the GPIO-16 port (when possible that is).

andyjbm, 2021-06-15:
Hi there,

Just jumped on the easyesp ship this week and I'm excited. So here I am with my observations using easyesp with DS18B20 in parasite power mode.

The problem is that in parasite power mode the one wire bus must be held high when a measurement is being performed or when the data is being written to eeprom. This is because of the amount of current required during this time. The DS18B20 cannot store enough energy to complete a conversion. It needs power from the bus during this time.

How it gets this power is not the debate for me - I've had success with really low pullup resistors (1k) when there are 8 or more DS18B20 or setting the GPIO pin as output high during conversion.

Now the crux of the problem with easyesp as I've just observed today.
With one DS18B20 the easyesp sends the 44 command code to initiate a conversion and then waits for 200mS with the data line high. I was using a 3k3 pullup to 3v3. Then it asks the DS18B20 for the results with command code BE.

This means that the DS18B20 gets the power it needs to achieve a successful conversion.

However with more than one DS18B20 easyesp sends the start measurement command (44) to the 1st DS18B20 **AND THEN IMMEDIATELY** sends a start measurement command to the next DS18B20 **starving the 1st of the power it needs** to succesfully perform a measurement. So the 1st fails and results in NaN but the last DS18B20 succeeds.

This was using one esyesp "device" in single vs dual vs quad mode.

I also had varying degrees of success by using each DS18B20 on its own easyesp "device" but on the same one-wire bus.

Conclusion: The issue with DS18B20 NaN is not communication or hardware issues but because the firmwaare implementation is not honouring the requirement of a DS18B20 to have the one-wire bus to be held high **WHILE A MEASUREMENT IS IN PROGRESS**. Easyesp is attempting to communicate with other devices on the one-wire bus and therefore **starving those DS18B20s that are trying to perform measurements of the power they need for succesful measurement.**

I had both a DS1054 and a picoscope looking at the onewire bus and both wave form and data are fine and not corrupt at all. The issue for most is not the pullup resistor or not likely to be wire length. It is the problem that the firmware is writing to the one-wire bus while a device is trying to perform a measurement.

Now how to fix? No idea yet. Only just got started with easyesp. Off to take a look at the code for the DS18B20 plugin.

Others that know the code better may beat me to a fix now they know the crux of the problem.

Regards,

Andy.

TD-er, 2021-06-15:
Parasitic powering of the Dallas sensors is not supported.
And the 'interleaving' of reading is done on purpose.
It does allow to get upto 4 sensors per task which will be read at the same time.
When using parasitic mode, this is impossible.
It also needs some synchronization between tasks to make sure the GPIO pin is not changed when another sensor is read.
So therefore I have chosen to not support parasitic mode... or at least not yet.

andyjbm, 2021-06-15:
@TD-er Ahh, Ok.

I misunderstood then in the device stats where it stated "Parasite Powered:" with a true flag as being recognised & therefore supported.

Maybe a comment on that page to advise that parasitic powered configuration is not supported would save those from the frustration of trying to get it to work and understanding why it is not working?

Is parasitic power config on the road map at all? I guess not in the immediate future from the wording of your reply...

Regards,  Andy.

andyjbm, 2021-06-15:
As an additional thought, I believe it is possible to initiate measurement/converrsion to all DS18b20 on the one-wire rather than addressing and initiating a measurement to each DS18B20 individually by sending a global 44h instead of a 55h and the ROM address and then 44h.

That should allow all devices in a quad to measure simultaneously and be read interleaved as is already the case but also preserving the state of DQ while measurement takes place?

The limitation would then me up to 4 DS18B20 per one-wire bus.

Andy.

andyjbm, 2021-06-15:
> As an additional thought, I believe it is possible to initiate measurement/converrsion to all DS18b20 simultaneously on the one-wire rather than addressing and initiating a measurement to each DS18B20 individually by sending a global 44h instead of a 55h and the ROM address and then 44h.
> 
> That should allow all devices in a quad to measure simultaneously and be read interleaved as is already the case but also preserving the state of DQ while measurement takes place?
> 
> The limitation would then be up to 4 DS18B20 per one-wire bus as a single easyesp "device".
> 
> Andy.

Nah, I'm not convinced I got this right. Can't find where I thought I saw it now. But it was a while ago.

A.

TD-er, 2021-06-15:
[See documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P004.html#number-output-values) where it is stated that parasitic power is not supported.

> As an additional thought, I believe it is possible to initiate measurement/converrsion to all DS18b20 simultaneously on the one-wire rather than addressing and initiating a measurement to each [...]

What happens with the Dallas sensors on several tasks is that they will start the measurement, one task after another and then reschedule the tasks to read the sensors.

However it is not guaranteed this will always be done in this sequence.
The (upto) 4 sensors in a task are read in the same sequence, so those will always be read at the same time.
The reason why it cannot be guaranteed for multiple tasks is that ESPEasy will run a lot more so things should not be blocking.
Starting an 1-wire measurement does take some time and the max. read period is 750 msec.
So I am not entirely sure more than 12 can be read at the same time. But this is already blocking for roughly a second to read them all at the same time.
1000 msec blocking is a big no-no as it will affect all other scheduled actions on the ESP.
You may even get into trouble with WiFi connection since the 1-wire bit toggling is rather blocking code which cannot be interrupted by checking WiFi in the mean time.

andyjbm, 2021-06-15:
Hi @TD-er ,

So I found the sequence to initiate a measurement on all Ds18B20 on the one wire bus in one hit. Perform a one-wire reset then write CCh (ROM skip command) followed by 44h (initiate measurement command). Then set the one-wire bus high.

This should syncronously start measurements on all the devices on the one wire bus. After the conversion time has passed the event schedule can then poll each device to read the values as it does already.

This has the downside of only working for 4 devices on the one bus and other easyesp "devices" with physical devices on the same bus would interfere with this process. Also other devices would be interfered with by this process!

I have another idea which must wait until I'm in front of a real keyboard to type it up!

Regards, Andy.

TD-er, 2021-06-15:
Just to avoid confusion in explaining.

In ESPEasy we have a plugin, which is a piece of code describing how to interact with some sensor (or other hardware like a display)
The items you see on the "Devices" tab are called "tasks".
A task is an instance of a plugin, with specific settings for a specific sensor (e.g. pin configuration, or hardware address, etc.)

So what you call "devices", you probably mean "tasks" right?

andyjbm, 2021-06-15:
> So what you call "devices", you probably mean "tasks" right?

Yes I think so, Not with the lingo yet...

A.

andyjbm, 2021-06-15:
Ok, So quick aside as I thought I'd got this nailed right now but as they say one step forward and you fall off the cliff...

So my use case for easyesp. I'd been using a PI to read in 10 dallas 10b20s and posting to a local emoncms server monitoring a heating system which has been recently ripped out and replaced. I need to monitor flow and return as a quick verification and I thought a wemos D1 mini and easyesp capable of both ds18B20 and emoncms would be plug and play....

However.

Parasite power not supported - fair enough. But it _does_ work if you only have one device on the bus. So I just set up two devices on separate GPIO. Fine and dandy. Values flowing out the serial monitor.

Now to emoncms that was working aok with NaN values across 4 sensors on one bus. As field1 field2 etc.
 
however now they are on separate "tasks" (yay!) both devices are referred to as "field1" rather than the task name or the task device value (correct notation?). And as the emoncms node id is tied to the easyesp instance on the wemos the two dallas values are pasting over the top of each other on one emoncms input instead of being sent to two different inputs (temp1, temp2) on the same node or the same field1 on two different nodes 41,42 etc.

Bugger, there's the cliff.

This "feature" is more to do with the emoncms controller than the dallas thread and ought to be raised somewhere else - probably would be nice to push the task value name to emoncms rather than field1 field2 etc. But it became apparent because of the problem I was having with the dallas DS18B20.

Bedtime.

Andy.

TD-er, 2021-06-15:
I have never used EmonCMS.
But if you need to send a task with 2 variables, why not store the values in a 'dummy' task and let the dummy task send its values to the EmonCMS controller?
From the rules you can store something in a dummy task using task [taskvalueset](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=dummy#taskvalueset)

To 'flush' the values in a dummy, you can call `taskrun`, which will send all values to the connected controller.

You can even call taskvaluesetandrun (as last one)

But why do you need to use the Dallas sensors in parasitic mode?
Why not adding an extra wire to them?

andyjbm, 2021-06-16:
> But why do you need to use the Dallas sensors in parasitic mode?
> Why not adding an extra wire to them?

The sensors are already in and wired. I was avoiding having to rewire them thinking this would work with the wemos as is. In hindsight (or starting afresh) a three wire one-wire (there's an irony) would have been the thing to do. And may well be what I end up doing in the end.

A.

andyjbm, 2021-06-16:
> I have never used EmonCMS.
> But if you need to send a task with 2 variables, why not store the values in a 'dummy' task and let the dummy task send its values to the EmonCMS controller?

So emoncms would take data in the form value1Name:value,value2Name:value etc. Easyesp emoncms plugin uses field1,field2 etc as the value name and this works when posting the data from a Ds18B20 task in quad mode. The one task sends the four values which get sent to emoncms as field1, field2, field3, field4. One for each temp sensor. Great.

However when using a separate plugin task for each Ds18B20, each task instance sends its own value via the emoncms controller as "field1", so the values from the four sensors end up in emoncms in one input feed instead of four separate ones.

> From the rules you can store something in a dummy task using task [taskvalueset](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=dummy#taskvalueset)
>
> To 'flush' the values in a [...]
I'm completely newbie to easyesp. It is obviously a capable bit of kit - my initial attraction to it being the WiFi setup out of the box and straight forward to get up & going on power efficient hardware.

I will have to set time aside to understand what it can do and give these other ways you describe a shot.

Thanks for the pointer...

A.

[10 more replies]

closed 2023-09-09 by TD-er