letscontrolit/ESPEasy #2746: Reset firmware to default page does not keep WIFI

fmuntean, 2019-11-16:
Using the latest Firmware build from mega branch
Had to reset the firmware defaults on my dev board ESP8266 4MB when not at home and selected to keep the WiFi settings however after reset the board did not keep those settings and asked me to select the WIFI network thus can't continue to the home page.

These are the logs I captured so far from the time I updated the WF:
Update: mfd_ESP8266_4096_80MHz_QIO_VCC_2019_11_16.bin
sleep disable
................................................................................................................................................................................................................................................................................................................................Update Success: 653392
Rebooting...

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
@cp:0
ld
�U9231 : Info  : 

INIT : Booting version: MFD-mega-20191116 (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
9232 : Info  : INIT : Free RAM:36104
9233 : Info  : INIT : Cold Boot - Restart Reason: Software/System restart
9235 : Info  : FS   : Mounting...
9256 : Info  : FS   : Mount successful, used 75802 bytes of 957314
9489 : Info  : CRC  : program checksum       ...OK
9520 : Info  : CRC  : SecuritySettings CRC   ...OK
9522 : Info  : CRC  : binary has changed since last save of Settings
9615 : Info  : INIT : Free RAM:33352
9616 : Info  : INIT : I2C
9616 : Info  : INIT : SPI not enabled
9624 : Info  : INFO : Plugins: 10 [Normal] (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
9625 : Info  : EVENT: System#Wake
9732 : Info  : WIFI : Set WiFi to STA
9764 : Info  : WIFI : Connecting Howard attempt #0
9778 : Info  : EVENT: System#Boot
11412 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 30088 WiFiStatus 6
12770 : Info  : EVENT: WiFi#Disconnected
12775 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2848 ms
12890 : Info  : WIFI : Connecting Howard attempt #1
15885 : Info  : EVENT: WiFi#Disconnected
15890 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2848 ms
16016 : Info  : WIFI : Connecting Howard attempt #2
19013 : Info  : EVENT: WiFi#Disconnected
19017 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2848 ms
19144 : Info  : WIFI : Connecting Howard attempt #3
22140 : Info  : EVENT: WiFi#Disconnected
22145 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2848 ms
22260 : Info  : WIFI : Connecting Howard attempt #4
25256 : Info  : EVENT: WiFi#Disconnected
25261 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2848 ms
25376 : Info  : WIFI : Connecting Howard attempt #5
28384 : Info  : EVENT: WiFi#Disconnected
28388 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2849 ms
28504 : Info  : WIFI : Connecting Howard attempt #6
29771 : Info  : WIFI : Set WiFi to AP+STA
29807 : Info  : EVENT: WiFi#APmodeEnabled
29811 : Info  : WIFI : AP Mode ssid will be ESP12-TEST with address 192.168.4.1
30135 : Info  : AP Mode: Client connected: 7C:5C:F8:50:10:CC Connected devices: 1
30138 : Info  : Webserver: start
41127 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 24848 WiFiStatus 0
60578 : Info  : EVENT: Clock#Time=Thu,00:01
61381 : Info  : SaveToFile: free stack: 3392
61405 : Info  : FILE : Saved config.dat
61406 : Info  : SaveToFile: free stack after: 3392
61411 : Info  : SaveToFile: free stack: 3312
61450 : Info  : FILE : Saved config.dat
61451 : Info  : SaveToFile: free stack after: 3312
61453 : Info  : SaveToFile: free stack: 3312
61474 : Info  : FILE : Saved security.dat
[... 146 lines cut ...]

INIT : Booting version: MFD-mega-20191116 (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
45 : Info  : INIT : Free RAM:35968
46 : Info  : INIT : Warm boot #2 Last Task: Const Interval timer, id: 2 - Restart Reason: Hardware Watchdog
49 : Info  : FS   : Mounting...
70 : Info  : FS   : Mount successful, used 75802 bytes of 957314
302 : Info  : CRC  : program checksum       ...OK
334 : Error : CRC  : SecuritySettings CRC   ...FAIL
406 : Info  : INIT : Free RAM:32808
409 : Info  : INIT : I2C
410 : Info  : INIT : SPI not enabled
419 : Info  : INFO : Plugins: 10 [Normal] (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
422 : Info  : EVENT: System#Wake
430 : Error : WIFI : No valid wifi settings
533 : Info  : WIFI : Set WiFi to AP
569 : Info  : EVENT: WiFi#APmodeEnabled
575 : Info  : WIFI : AP Mode ssid will be ESP12-TEST_0 with address 192.168.4.1
578 : Error : WIFI : Could not prepare WiFi!
594 : Info  : EVENT: System#Boot
2228 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 28560 WiFiStatus 6
32208 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 28528 WiFiStatus 6
60690 : Info  : EVENT: Clock#Time=Thu,00:01
62306 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 28512 WiFiStatus 6
92273 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 28512 WiFiStatus 6
120741 : Info  : EVENT: Clock#Time=Thu,00:02

uzi18, 2019-11-16:
confirmed, it wasn't work for me from the beginning

fmuntean, 2019-11-16:
When I reset firmware I selected all the checkboxes to keep the settings.

From what I can see the culprit is:
333 : Error : CRC : SecuritySettings CRC ...FAIL 

I also see other issues like:
- watchdog resetting the board once the client disconnected.
- WIFI appending the unit number to the name even if before that was unselected. (I expect that option to be kept with the unit name)

TD-er, 2019-11-16:
The CRC is not the issue.
Watchdog is a very likely cause.

Also the reset may be taken place before the caches for SPIFFS are flushed.

Sprinterfreak, 2020-04-14:
I also struggle with an unexpected factory reset bahavior.
If i reset my device once, it always drops into APMode instead of using WiFi settings from compile time (Custom.h). This causes the device to be completly loose its connectivity.

Reseting it again, of course only possible via UART, brings back settings from compile time (Custom.h).
Even more unsatisfying is that my devices are, because of a flash hardware-upgrade, not recognized by the DeviceModel. So no reset button works. (See PR #3018 )

Maybe related.

commit f77b2e18 (2023-08-07): [Factory Reset] Fix keeping settings (#4263)

Fixes: #4263
Fixes: #2746
Fixes: #2226
closed 2023-08-07 by TD-er