letscontrolit/ESPEasy #2749: SendTo/SendToHttp not working in  20191116 built

jimikid2002, 2019-11-17:
Tried both new engine and old engine, SendTo and SendToHttp through "too many argument" error and the action is not working. Downgraded to 20190925 build and it is working.

TD-er, 2019-11-17:
Then your rules must have a very small change.
See https://github.com/letscontrolit/ESPEasy/issues/2724

If you need help, please post your rules line.
If you have it working again, please close this issue.

jimikid2002, 2019-11-17:
On intru#intru=1 do
gpio,4,1
SendTo 0,event,alarm11
endon

On rst do
gpio,4,0
SendTo 0,event,alarm10
endon

TD-er, 2019-11-17:
```
On intru#intru=1 do
gpio,4,1
SendTo 0,"event,alarm11"
endon

On rst do
gpio,4,0
SendTo 0,"event,alarm10"
endon
```

I added the two quotes for each `SendTo` line.

jimikid2002, 2019-11-17:
Dear Noorlander,
Thanks for your response, I have tried your suggested syntex, now the error message is gone but the event is not reaching unit 0.

[image]
Regards
Jimi

Grovkillen, 2019-11-17:
If you do not use ZERO as the unit number of the recieving unit? FYI: zero is not a valid unit number.

jimikid2002, 2019-11-18:
> If you do not use ZERO as the unit number of the recieving unit? FYI: zero is not a valid unit number.

Dear Grovkillen,

I used unit 0 in other projects and it really works. Older releases (20190925) working perfectly in this project as well.  

I double checked before replying.... :-)

Regards
Jimi

TD-er, 2019-11-18:
> I used unit 0 in other projects and it really works. Older releases (20190925) working perfectly in this project as well.

But have you checked now not to use unit 0 for the ESPEasy p2p protocol?
I recently changed some inconsistencies in the code and now things will be enforced.
So unit 0 and unit 255 should not be used for our p2p protocol.

jimikid2002, 2019-11-18:
I will try that and share the result. Thanks.

jimikid2002, 2019-11-18:
YES!!!!!
After changing unit 0 to unit 10 it is working....... Thanks a lot.
[image]
****

TD-er, 2019-11-18:
Thanks for letting us know.
Will close the issue right now.

closed 2019-11-18 by TD-er