letscontrolit/ESPEasy #2750: ESP32 serial server doesn't work.

1igeza, 2019-11-17:
After enabling serial server on esp32 and reboot it, the device won't boot again.
Tried on two different esp32.  esp32test_4M316k
```
INIT : Booting version:  (ESP32 SDK v3.2.3)
85 : INIT : Free RAM:291476
85 : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
91 : FS   : Mounting...
139 : CRC  : No program memory checksum found. Check output of crc2.py
153 : CRC  : SecuritySettings CRC   ...OK 
243 : INIT : Free RAM:285824
245 : INIT : SPI Init (without CS)
Backtrace: 0x4008db14:0x3ffb19e0 0x4008dd45:0x3ffb1a00 0x4008a649:0x3ffb1a20 0x401bb546:0x3ffb1a60 0x401bb822:0x3ffb1a80 0x401ab384:0x3ffb1aa0 0x401ab3ed:0x3ffb1ac0 0x401bd062:0x3ffb1ae0 0x401bbd1c:0x3ffb1b00 0x401a8c00:0x3ffb1b40 0x400d6926:0x3ffb1b60 0x4015a7d6:0x3ffb1ba0 0x4010c2aa:0x3ffb1dc0 0x4010d9ca:0x3ffb1e80 0x4015eca6:0x3ffb1ee0 0x40175e17:0x3ffb1fb0 0x40089a49:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5788
entry 0x4008069c
�ff�U���2-hal-cpu.c:159] setCpuFrequencyMhz(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.
�U84 :  
```

TD-er, 2019-11-17:
Hmm, if it does cause a lot of crashes, it should in the end disable the task to be able to boot again.

Does that happen? After (+/- 10 + taskindex) reboots it should be disabled.

I have not tested the serial server on ESP32 myself.

1igeza, 2019-11-17:
It won't boot again even after 50 reboots, I have to clear flash and upload espeasy again to esp32 to make it work again.

TD-er, 2019-11-17:
It should be possible to reset to factory default by connecting RX and TX at boot.

1igeza, 2019-11-18:
Reset also doesn't work with RX and TX connected at boot. 
```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5788
entry 0x4008069c
⸮U49 : Info  : 

INIT : Booting version:  (ESP32 SDK v3.2.3)
49 : Info  : INIT : Free RAM:291556
50 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
58 : Info  : FS   : Mounting...
89 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
105 : Info  : CRC  : SecuritySettings CRC   ...OK 
170 : Info  : INIT : Free RAM:285868
171 : Info  : INIT : I2C
171 : Info  : INIT : SPI not enabled
350 : Info  : INFO : Plugins: 75 [Normal] [TEST ESP32] (ESP32 SDK v3.2.3)
451 : Info  : WIFI : Set WiFi to STA
565 : Info  : WIFI : Connecting Rafal attempt #0
571 : Info  : OTA  : Arduino OTA enabled on port 3232
820 : Info  : SYS  : 0.00,0.00
>170 : Info  : INIT : Free RAM:285868
833 : Info  : Command: 170
833 : Info  : Command unknown: "170"

Unknown command!
>171 : Info  : INIT : I2C
957 : Info  : Command: 171
957 : Info  : Command unknown: "171"

Unknown command!
>171 : Info  : INIT : SPI not enabled
1081 : Info  : Command: 171
1081 : Info  : Command unknown: "171"

Unknown command!
>350 : Info  : INFO : Plugins: 75 [Normal] [TEST ESP32] (ESP32 SDK v3.2.3)
1204 : Info  : Command: 350
1204 : Info  : Command unknown: "350"

Unknown command!
>451 : Info  : WIFI : Set WiFi to STA
1213 : Info  : Command: 451
1214 : Info  : Command unknown: "451"

Unknown command!
>565 : Info  : WIFI : Connecting Rafal 820
2037 : Info  : Command: 565
2038 : Info  : Command unknown: "565"

Unknown command!
>>170 : Info  : INIT : Free RAM:28588>171 : Info  : INIT : I2C9>171 : Info  : INIT : SPI not enabled1>350 : Info  : INFO : Plugin
2053 : Info  : Command: >170
2054 : Info  : Command unknown: ">170"

Unknown command!
2156 : Info  : WIFI : Connected! AP: Rafal (B0:6E:BF:32:33:30) Ch: 0 Duration: 1487 ms
2158 : Info  : WIFI : DHCP IP: 192.168.5.73 (TEST-ESP32-0) GW: 192.168.5.1 SN: 255.255.255.0   duration: 24 ms
2167 : Info  : Webserver: start
2168 : Info  : firstLoopConnectionsEstablished
>1204 : Inf>565 : Info  : WIFI : Connecting Rafal 8202037 : Info  : Command: 565
2185 : Info  : Command: 1204
2185 : Info  : Command unknown: "1204"

```

uzi18, 2019-11-18:
it shows it work ok after this reboot, you got IP and connection

1igeza, 2019-11-18:
I didn't configured serial server, just setup system info as device just for testing resetting by connection tx and rx. If it would work, there should be nothing in device tab, and WiFi should works as AP with 192.168.4.1.  I run esp32 for about 2 minutes with rx + tx, then unconnected this pins and reboot esp32 and nothing was delete in device tab.

TD-er, 2019-11-18:
No, you should connect the pins and then reboot the ESP.
There is a check in the code to see if it will read the same specific data as it does send.
If so, then there is a short detected at boot and it will perform a factory reset.

1igeza, 2019-11-18:
I did exactly what you said, Connect rx to tx, power on, wait about 2 minutes, power off, unconnect rx and tx, power on. I though that it is obvious that I rebooted the device after connecting this two pins.  (I posted log with connected rx and tx while booting few posts before)

uzi18, 2019-11-18:
in fact serial connection on esp32 is not well tested (as many other features) , so there could be some malfunction

rbakker69, 2020-02-07:
I have the same issue. When configuring serial server, it directly receive serial events, but after reboot something goes wrong and a stack trace is showed. Can I somehow help to get this fixed? I want to use the ESP32 Ser2Net function for RFlink and some oter ESP functions (e.g. display en touch button).

tlc76, 2020-09-20:
Hello,

I have a similar problem.
Device : ESP32 Doit Devkit v1 board, with ESP_Easy_mega_20200913_test_ESP32_4M316k binary flashed.

Everything works perfectly until I create a Serial Server and enable it.
Once I do that, after the first reboot the device is no longer booting.
If I keep the Serial Server disabled, the ESP32 boots correctly.
[image]

As soon as I enable the Serial Server and reboot the ESP32, it will no longer boot.
I get this infinite loop:

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
▒U59 : Info  :

INIT : Booting version:  (ESP32 SDK v3.2.3-14-gd3e562907)
60 : Info  : INIT : Free RAM:291408
60 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
60 : Info  : FS   : Mounting...
99 : Info  : CRC  : SecuritySettings CRC   ...OK
170 : Info  : INIT : Free RAM:286560
170 : Info  : INIT : I2C
171 : Info  : INIT : SPI not enabled
Backtrace: 0x4008dc38:0x3ffb1980 0x4008de69:0x3ffb19a0 0x40089fed:0x3ffb19c0 0x401f114a:0x3ffb1a00 0x401f1426:0x3ffb1a20 0x401dfb30:0x3ffb1a40 0x401dfb99:0x3ffb1a60 0x401db5f2:0x3ffb1a80 0x401da01c:0x3ffb1aa0 0x401dea94:0x3ffb1ae0 0x400d7e3a:0x3ffb1b00 0x40148fca:0x3ffb1b40 0x40103591:0x3ffb1d70 0x40108a76:0x3ffb1e70 0x401502b6:0x3ffb1ed0 0x401836f7:0x3ffb1fb0 0x4008a301:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
▒U59 : Info  :

INIT : Booting version:  (ESP32 SDK v3.2.3-14-gd3e562907)
60 : Info  : INIT : Free RAM:291408
60 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
60 : Info  : FS   : Mounting...
99 : Info  : CRC  : SecuritySettings CRC   ...OK
169 : Info  : INIT : Free RAM:286560
170 : Info  : INIT : I2C
171 : Info  : INIT : SPI not enabled
Backtrace: 0x4008dc38:0x3ffb1980 0x4008de69:0x3ffb19a0 0x40089fed:0x3ffb19c0 0x401f114a:0x3ffb1a00 0x401f1426:0x3ffb1a20 0x401dfb30:0x3ffb1a40 0x401dfb99:0x3ffb1a60 0x401db5f2:0x3ffb1a80 0x401da01c:0x3ffb1aa0 0x401dea94:0x3ffb1ae0 0x400d7e3a:0x3ffb1b00 0x40148fca:0x3ffb1b40 0x40103591:0x3ffb1d70 0x40108a76:0x3ffb1e70 0x401502b6:0x3ffb1ed0 0x401836f7:0x3ffb1fb0 0x4008a301:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c`

I connected the physical device (CC2530+CC2591 zigbee gateway) to the TX2 and RX2 pins, but the problem is that I am unable to select these pins in the Serial Server configuration page!
I believe the problem is caused by the fact that by default, the Espeasy firmware is using TX0/RX0 UART pins which are used at boot. This permanently crashes the device, making it un-bootable. The only way to solve the proble is to reflash it.

The TX0/RX0 pins are directly connected to the ESP32 memory, which causes the crash at boot.
Practically the Serial Server functionality is not usable on ESP32 devices (with the default pin configuration).

How can I change the default serial setting to use other pins and not the default ones (TX0/RX0)?
This problem is NOT impacting the ESP8266 devices.

Thanks,
Cristian

TD-er, 2020-09-20:
Hmm it looks like this plugin is not using the ESPEasy Serial wrapper as all other plugins do.
This makes it impossible to select a different UART port.
Until that's added, this will not work on an ESP32.

tlc76, 2020-09-21:
Thanks for the answer.
I found the sources for the ser2net plugin (file _P020_Ser2Net.ino) but I am not sure if the UART pins are set here in this file or somewhere else.
Can you kindly advise how to change the RX/TX pin configuration in the EspEasy sources?
If I manage to change these pins to RX2/TX2 then recompile, that should solve the problem I guess.

Cheers

TD-er, 2020-09-21:
That plugin is probably using `Serial` which is an alias for `Serial0`
So if you change it to Serial1, you already do use a different serial port and you can also assign the right pins for it on that C++ object
Just make sure you don't use Serial0 as you're doing now as that does cause the current issues you're now seeing.

tlc76, 2020-09-22:
I changed all Serial. to Serial2. in file _P020_Ser2Net.ino
For example: 
line
Serial.begin(ExtraTaskSettings.TaskDevicePluginConfigLong[1], (SerialConfig)serialconfig);

becomes
Serial**2**.begin(ExtraTaskSettings.TaskDevicePluginConfigLong[1], (SerialConfig)serialconfig);

In the end I have the exact same behavior.

Then I changed the ESP32 pin definitions in ESPeasySerial.h, to have all 3 UARTs pointing at pins 16 and 17.
It didn't work, the boot loop persists...

> #ifndef ESP32_SER0_TX
>   # define ESP32_SER0_TX 17
> #endif // ifndef ESP32_SER0_TX
> #ifndef ESP32_SER0_RX
>   # define ESP32_SER0_RX 16
> #endif // ifndef ESP32_SER0_RX
> 
> #ifndef ESP32_SER1_TX
>   # define ESP32_SER1_TX 17
> #endif // ifndef ESP32_SER1_TX
> #ifndef ESP32_SER1_RX
>   # define ESP32_SER1_RX 16
> #endif // ifndef ESP32_SER1_RX
> 
> #ifndef ESP32_SER2_TX
>   # define ESP32_SER2_TX 17
> #endif // ifndef ESP32_SER2_TX
> #ifndef ESP32_SER2_RX
>   # define ESP32_SER2_RX 16
> #endif // ifndef ESP32_SER2_RX

Any ideas?...
@TD-er , can you please advise how to assign the right pins for it on that C++ object (I assume you were referring to the Serial object) ?

Thanks a lot,
Cristian

TD-er, 2020-09-22:
`ESPeasySerial.h` is not (yet) used for this plugin, so no need to change stuff there
What you need to do is to set the GPIO pins for the Serial2 port (or Serial1) and use that one. (that part you already claim to have done)

tlc76, 2020-09-23:
I am not sure I understand correctly where does this plugin stores the UART pin configuration.
Per the EspEasy Development wiki (https://letscontrolit.com/wiki/index.php/ESPEasyDevelopment):

> Plugin configuration
> Basic configuration is already provided by the framework, like names, GPIO selection, port selection, etc. So maybe you don't have to worry about this at all. You just have to read the required variables if you need them:
```
Settings.TaskDevicePin1[event->TaskIndex] - The first GPIO pin selected within the task
Settings.TaskDevicePin2[event->TaskIndex] - The second GPIO pin selected within the task
Settings.TaskDevicePort[event->TaskIndex] - The port in case the device has multiple in/out pins
```

Are the TX/RX pins stored here? :
Settings.TaskDevicePin1[event->TaskIndex]
Settings.TaskDevicePin2[event->TaskIndex]
Somehow I doubt it, because as far as I can see in the sources, the TaskDevicePin1 and 2 are used to store the "Reset target after boot" and "TX Enable Pin" pins...
```
      addFormPinSelect(F("TX Enable Pin"),           F("taskdevicepin2"), Settings.TaskDevicePin2[event->TaskIndex]);

      addFormPinSelect(F("Reset target after boot"), F("taskdevicepin1"), Settings.TaskDevicePin1[event->TaskIndex]);
```
If these are not the TX/RX pins, can you kindly advise how to set these pins in the ser2net plugin (_P020_Ser2Net.ino) ?

Thanks for the help!
And apologies for the (possibly) stupid questions...

Best,
Cristian

rbakker69, 2020-09-27:
Some other thoughts for solving this issue:

When using Arduino and upload a example sketch with using the serial port (rx0/tx0) everything works and after a power cycle the sketch keeps working using the default serial port. The serial port shows first some boot entries and then switch over to the sketch output.

As the serial server module in ESPeasy does works fine and only gives a problem on reboot or power cycle, could a timing issue causing the problem? Is it possible to delay the initialization of serial server module, as the serial server module works just fine, after enabling it manualy?

Of course would it also be nice to be able to select another serial port and using two serial server modules with different TCP ports, as the ESP32 has three serial ports. But for now, it think it should help to just delay the initialization of the module.

tlc76, 2020-09-30:
I can confirm that the Serial Server works fine on default serial port (rx0/tx0), once configured and enabled, until the first reboot.
Indeed the issue appears during reboot at the initialization of plugins:

> INIT : Booting version: (ESP32 SDK v3.2.3-14-gd3e562907)
> 60 : Info : INIT : Free RAM:291408
> 60 : Info : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 60 : Info [...]

Any help is highly appreciated.

Regards

tlc76, 2020-12-23:
Meanwhile I managed to decode the exception.
The test scenario is the following:
- ESP32 board: DevKit v1 (30 pin version)
- ESP32 board NOT physically connected to any other board via serial or other pins
- ESP32 flashed with latest development branch dating from 23-Dec-2020
- File Custom.h : enabled #define USES_P020   // Ser2Net
- Flashed ESP_Easy_mega_20201224_custom_ESP32_4M316k-factory.bin (freshly compiled) file
- Added new task/device:
[image]
- Rebooted the ESP32
- Entered the boot loop problem.

The new backtrace info is:
```
232574 : Info   : EVENT: TaskInit#test=1,1
232673 : Info   : EVENT: !Serial#▒
242768 : Info   : EVENT: Clock#Time=Thu,00:10
243323 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 228172 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
261944 : Info   :  Webserver args: 0: 'cmd' length: 6
261944 : Info   :      : Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
▒U28 : Info   :

INIT : Booting version: Brick4U: Dec 24 2020 01:00:33 (ESP32 SDK v3.2.3-14-gd3e562907)
29 : Info   : INIT : Free RAM:297304
30 : Info   : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
30 : Info   : FS   : Mounting...
72 : Info   : CRC  : SecuritySettings CRC   ...OK
147 : Info   : INIT : Free RAM:292384
148 : Info   : INIT : I2C
148 : Info   : INIT :
Backtrace: 0x4008d5f8:0x3ffb19e0 0x4008d829:0x3ffb1a00 0x40089831:0x3ffb1a20 0x401b85c2:0x3ffb1a60 0x401b889e:0x3ffb1a80 0x401a6fa8:0x3ffb1aa0 0x401a7011:0x3ffb1ac0 0x401a2a6a:0x3ffb1ae0 0x401a1494:0x3ffb1b00 0x401a5f0c:0x3ffb1b40 0x400d77d2:0x3ffb1b60 0x400f9c06:0x3ffb1ba0 0x4010c00f:0x3ffb1dd0 0x400f300d:0x3ffb1eb0 0x400f8d4b:0x3ffb1ee0 0x401513e7:0x3ffb1fb0 0x40089b45:0x3ffb1fd0

Rebooting...
```

I decoded the above info using the esp32-backtrace tool and the ESP_Easy_mega_20201224_custom_ESP32_4M316k.elf file generated during compilation by VSCode+PlatformIO.
Here are the results:

```
BT-0: 0x4008d5f8 is in invoke_abort (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:155).
BT-1: 0x4008d829 is in abort (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:170).
BT-2: 0x40089831 is in xQueueGenericReceive (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442).
BT-3: 0x401b85c2 is in sys_mutex_lock (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:78).
BT-4: 0x401b889e is in sys_arch_protect (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:469).
BT-5: 0x401a6fa8 is in do_memp_malloc_pool (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c:302).
BT-6: 0x401a7011 is in memp_malloc (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c:398).
BT-7: 0x401a2a6a is in netconn_alloc (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c:742).
BT-8: 0x401a1494 is in netconn_new_with_proto_and_callback (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c:133).
BT-9: 0x401a5f0c is in lwip_socket (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:1587).
BT-10: 0x400d77d2 is in WiFiServer::begin(unsigned short) (D:/Date/kit/VSC_PlatformIO_Python/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/sockets.h:593).
BT-11: 0x400f9c06 is in Plugin_020(unsigned char, EventStruct*, String&) (D:/Date/kit/espeasy/ESPEasy_ESP32_mega-development/src/_P020_Ser2Net.ino:130).
BT-12: 0x4010c00f is in PluginCall(unsigned char, EventStruct*, String&) (src\src\Globals/Plugins.cpp:462).
BT-13: 0x400f300d is in PluginInit() (D:/Date/kit/espeasy/ESPEasy_ESP32_mega-development/src/__Plugin.ino:1060).
BT-14: 0x400f8d4b is in setup() (D:/Date/kit/espeasy/ESPEasy_ESP32_mega-development/src/ESPEasy.ino:396).
BT-15: 0x401513e7 is in loopTask(void*) (D:\Date\kit\VSC_PlatformIO_Python\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:14).
BT-16: 0x40089b45 is in vPortTaskWrapper (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143).
```

Looking into the corresponding source files for some of the decoded backtrace information, I noticed that the problem is caused at the plugin initialization:
_P020_Ser2Net.ino line 130:        ser2netServer->begin();
__Plugin.ino line 1060:   PluginCall(PLUGIN_INIT_ALL, nullptr, dummy);
ESPEasy.ino line 396: PluginInit();

I am attaching some useful files:
- sources:
[ESPEasy-mega_20201224 - Copy.zip](https://github.com/letscontrolit/ESPEasy/files/5737459/ESPEasy-mega_20201224.-.Copy.zip)
 
- elf file:
[debug_elf_file.zip](https://github.com/letscontrolit/ESPEasy/files/5737461/debug_elf_file.zip)

Is there anyone able to help here by identifying the cause of the issue?
The bug is quite nasty because it makes the use of EspEasy Serial Server plugin unusable.

Let me know in case I can help with any tests or debug information.

many thanks

TD-er, 2020-12-23:
Will mark this post as unread, so I can have a look at it later.

tlc76, 2021-01-14:
Hi @TD-er, sorry to jump in : have you managed to look into this issue?
Thank you!

TD-er, 2021-01-14:
Not yet :(

uzi18, 2021-01-14:
@TD-er @tlc76 @1igeza @rbakker69 it is socket allocation error, looks like broken esp-idf api.

Please try if same problem you have with P1GatewayServer plugin.

immiimmi, 2021-01-17:
Hi uzi18
tested on esp32 
i can confirm the problem affects only P020_Ser2net
P044_P1WifiGateway does not crash the esp32 after reboot 
immi

TD-er, 2021-01-17:
I think it is related to this PR/issue on ESP32: https://github.com/espressif/arduino-esp32/commit/c6a8da61f78452573df857cb95edb3a5ecd0365f

Can you try with ESP32 core lib 2.1.0?
See: https://github.com/letscontrolit/ESPEasy/blob/cdc8a1a84c85c18266b75144bac6c156d3ddfa6c/platformio_core_defs.ini#L145

Just change the `2.0.0` into `2.1.0` and perform a clean rebuild.

uzi18, 2021-01-17:
please try this one: 
[ESP_Easy_mega_20210117_custom_ESP32_4M316k.bin.zip](https://github.com/letscontrolit/ESPEasy/files/5826595/ESP_Easy_mega_20210117_custom_ESP32_4M316k.bin.zip)

uzi18, 2021-01-17:
@TD-er P1 plugin does not execute begin on wifiserver when wifi is not connected

TD-er, 2021-01-17:
> @TD-er P1 plugin does not execute begin on wifiserver when wifi is not connected

"Does not" or "should not" ?

uzi18, 2021-01-17:
Does not, this is a difference 020 vs 044

immiimmi, 2021-01-31:
Dear espeasy experts; 
please find enclosed an updated ser2net; 
it would be great if you could review it and commit it as standard ser2net;
- no more  crash after reboot: tested positively with espressif32@2.0.0 and espressif32@2.1.0 
- can use all serials avaliable on the esp; not only serial0
- can use 2 serial2net at the same time: tested on esp32 with serial0 on port 1234 and serial2 on port 4321
- probably also 3 ser2net are possible; not tested 
b.r.
immi

p.s. all 3 files are needed
src/PluginStructs/P020_data_struct.h
src/PluginStructs/P020_data_struct.cpp
src/_P020_Ser2Net.ino
[P20_immi.zip](https://github.com/letscontrolit/ESPEasy/files/5900240/P20_immi.zip)

TD-er, 2021-01-31:
I made a PR from your ZIP file: https://github.com/letscontrolit/ESPEasy/pull/3483
It looks like there are some actual changes in the changed code, so it is a breaking change if I merge it like this.