letscontrolit/ESPEasy #2751: P052 Senseair plugin and depp dleep issue

sobolkz, 2019-11-18:
According to the results of #2616 and #2636 - same issue with P052 Senseair plugin was reproduced in mega-20191113 and mega-20191116 - after board was going back from deep sleep it was stopping boot proccess after Senseair S8 was detected and going back to sleep:

> INIT : Booting version: mega-20191116 (ESP82xx Core 2_6_0, NONOS SDK 2.2.2-dev(bb83b9b), LWIP: 2.1.2 PUYA support)
> 83 : Info  : INIT : Free RAM:30744
> 84 : Info  : INIT : Rebooted from deepsleep #4 Last Task: Const Interval timer, id: 1 - Restart Reason: Deep-Sleep Wake
> 86 : Info  : FS   : Mounting...
> 111 : Info  : FS   : Mount successful, used 76304 bytes of 957314
> 559 : Info  : CRC  : program checksum       ...OK
> 594 : Info  : CRC  : SecuritySettings CRC   ...OK
> 701 : Info  : INIT : Free RAM:27568
> 702 : Info  : INIT : I2C
> 703 : Info  : INIT : SPI not enabled
> 898 : Info  : PMSx003: using software serial
> 3983 : Info  : S/N: 72b56eb - Type: 10e -
> 8753 : Info  : INFO : Plugins: 80 [Normal] [Testing] [Development] (ESP82xx Core 2_6_0, NONOS SDK 2.2.2-dev(bb83b9b), LWIP: 2.1.2 PUYA support) ### - going back to deep sleep after it.

But instead old issue this bootloop can be ended if sensor was disconnected from board.
It's look like some timeout in version of WiFi plugin (because board going sleep before WiFi was connected).

TD-er, 2019-11-18:
What is the system awake time here?

It looks like the initialization of the S8 (I assume) sensor may block the system for a while which may prevent the WiFi from connecting within the set time.

sobolkz, 2019-11-18:
> What is the system awake time here?
> 
> It looks like the initialization of the S8 (I assume) sensor may block the system for a while which may prevent the WiFi from connecting within the set time.

Basically it was checked at 90 awakening\660 sleeping cycle with both newest releases. But also it was checked with 30\30 cycle.
All test was maked for both last fw versions, bootloop after deep sleep with connected S8. Each failed test - device wasn't connected to WiFi before going sleep.

sobolkz, 2020-01-04:
Hi all and happy New Year!
I have some free time and work with latest mega-20191208, it's look like no issue with P052 plugin in this version. Tested for about 3 days - no problems with deep sleep and awakening. Still looking for any other issues with my board.

closed 2020-01-04 by sobolkz