letscontrolit/ESPEasy #2754: New GPIO handling - Work in progress

giig1967g, 2019-11-19:
I am referencing the conversation of #2738:

Ok. Anyway, I was just trying to fix a bug in the longpulse command that currently is not working.

In order to find a solution for the bug, I have looked at the code again and come up with another idea. But before starting coding it I need your approval:

I noticed that there is a GPIO_TIMER.
What if:
1) I put all the read/write functions of P001, P009 and P019 in a separate module outside the 3 plugin 
2) I then use the GPIO_TIMER from the longpulse command that will call directly the read/write functions
3) In this way, we start putting in the core the GPIO basic handling, and gradually I could be moving out the single commands.

If you find it useful, I can start coding it.

_Originally posted by @giig1967g in https://github.com/letscontrolit/ESPEasy/pull/2738#issuecomment-554267552_

TD-er, 2019-11-19:
I think starting with the first point is already quite a bit of work.

> I put all the read/write functions of P001, P009 and P019 in a separate module outside the 3 plugin

giig1967g, 2019-11-19:
@TD-er 
So far I have a draft working version of the new logic.
1) done and working
2) done and working
3) so far done only for GPIOs commands (gpio, pcfgpio,mcpgpio) and working

After moving the GPIO command from plugin 1, 9 and 19 to a general module, I am facing some questions that I need to check with you:

a) **Command name:**
shall we keep using different gpio commands (`gpio`, `pcfpgio`, `mcpgpio`) or shall we modify the GPIO command to a new one that takes as a parameter the port/expander type? Example: `gpio,pcf,65,1` or `gpio,internal,13,1`
This will be a breaking change!
Another solution could be to keep the existing commands (marking them as deprecated in the documentation and in the log) and create a new generic one, like:
`setpin,internal,13,1`
`setpin,pcf,65,1`

b) **Events generated**:
as the new GPIO command is outside the plugin, there is no more connection with tasks.
Also, the GPIO command is supposed to work only with OUTPUT pins.
So, which types of events shall we send?
Currently if you define the GPIO in the task list, you get the event in the format `[task#value]=1`, and if you run the `monitor,gpio,13` command, you also get the `GPIO#13=1` event.
In my opinion, as the commands are not part of the plugin anymore, you should get only the `GPIO#13=1` event. And not the `[task#value]=1` event.
The reason to keep only one, is that we might have two events for the same pin change, and that is wrong in my opinion.

c) in the pinstate structure (GlobalMapPortStatus) the unique key is created using PLUGIN_ID and port number. Is it ok if for now we use in the core functions the original pluginID? In other words, we will use 1 for internal GPIO, 9 for MCP ports and 19 for PCF ports. In this case we will see in the pinstate page, under the same plugin number, both input and output pins. What do you think?

TD-er, 2019-11-19:
### Command names:
I think we should use the new ones, although we should be clear on how the pins will be named.
Maybe add a port start number in the plugins for now? Then we can use the same gpio command for all and the number defines what expander to use.
It is not likely there will be a mix of expander types.

To make the change a bit less breaking, we can implement the old command structure and "convert" the command to the new structure.
We should make the old ones deprecated and log this when used.

I think the `setpin` is a great name, along with `getpin`, since it makes the direction clear.

### Events
I'm not sure we should only act on OUTPUT pins.
We should be able to register a pin like you did in the past, but then register it for a specific event.
For example the ESP32 can trigger on rise/fall/change and has its own debounce in hardware.
For ESP8266 we must do it in software, but still from a monitoring perspective there should be no difference.
All of these monitored actions should be able to send out an event.

I am not sure yet what the event should be like.
As it is the result of monitoring, should it be indicated in the event name like "monitor"?
Maybe the monitor should be configurable to tell what should happen when an action is triggered?
For example to call a specific task?

The options are (not complete):
- Send event (what name? what parameters?)
- Call a task (what parameters?)
- Send a command (definable per pin?)

I think the event name structure of `GPIO#13=1` does make sense.
We somehow need to have some defined mapping from GPIO number to pin on an expander.
Maybe we can add some user defined name for it, to make stuff readable and the rules smaller. (you must trigger on `GPIO` otherwise)
So in the pin-trigger setup we should have some user defined string like `lightswitch` which will then be sent out instead of `GPIO#13`.
But maybe that's for future improvements.

### GlobalMapPortStatus
I think we should somewhere make a filter to see where a pin event is coming from or being sent to.
The most important thing is that this lookup can be fast and that we allow to have multiple GPIO expanders (that will be a request in the future, for sure).
So having a bit map to indicate what port expander to use makes sense.
Not sure if 1, 9 and 19 are the most useful then, but since it will be a run time value, I guess you could start with it. (will not be stored)

giig1967g, 2019-11-19:
**Command names:**
so I will leave both old `gpio`, `pcfgpio `and `mcpgpio `and mark them as deprecated and create a new one called `setpin` i.e. `setpin,internal,13,1`, `setpin,pcf,65,1`, `setpin,mcp,1,1`.

**Events**
I will explain better. 
Commands like gpio, longpulse, gpiotoggle, pwm, pulse, inputswitchstate(?) and tone can ONLY work on output pins as you cannot set or change an input pin! Infact if you use one of those commands on an INPUT pin the command will change the pinmode to OUTPUT.
We have to remember that the P001/009/019 plugins are only for INPUT pins (as the name suggests: Switch Input). 
Commands like monitor, status, unmonitor can work on both INPUT and OUTPUT pins.
The question is: for OUTPUT pins (that should not be declared in the P001/009/019 plugins) do we need the event [task#value] or it's enough the event `gpio#13=1`. I think that event [task#value] is enough.

And maybe create a new one for OUTPUT pins (name could be: Output Control).

TD-er, 2019-11-19:
> setpin,internal,13,1, setpin,pcf,65,1, setpin,mcp,1,1.

Maybe even consider the extra pins enumerated further?
Then you don't need to have `internal` or `pcf` defined in the command.
Only thing is, the pin count for ESP32 goes beyond ESP8266's pincount.

OK, so those commands like PWM or tone could later maybe also made useful for PWM chips?
And what about servo control? That's now also part of this list.

About the IO expanders.
I just checked and at least the MCP23017 is bi-directional.
So it can act as an input and an output.
Not sure if that's supported right now, but we can keep it in mind.

giig1967g, 2019-11-19:
your suggestion is to:
`setpin,[1-99],value` for internal
`setpin,[101-299],value` for mcp
`setpin,[301-499],value` for pcf

Isn't it a bit complicated to explain? Especially during the period in which both command (mcpgpio and setpin) will survive? All the documentation must be changed in order to indicate a different port number convention according to the command used. See pcf example:
[image]

I can confirm that both expanders are bidirectional (PCF and MCP).

TD-er, 2019-11-19:
Hmm if the expanders do have their own port range well defined, then there is no need to make it more complex and I would go with your suggestion.
My idea was to enumerate in a single row of numbers, but that may not be that useful since you cannot exchange rules among nodes with different pins (e.g. ESP32/8266)

So go with your idea to have a source/destination.
Maybe we should have it as last argument. If the port number is guaranteed to be unique (e.g. pin 101 is clear what it will be), then there is no need for the surplus target to be mentioned.

giig1967g, 2019-11-19:
It's no guaranteed to be unique:
MCP and PCF have same port number range (1-128):
[image]

giig1967g, 2019-11-19:
the idea of creating a single `setpin `command is more for a cosmetic reason.
we can also have:
- togglepin,internal,
- pulsepin, pcf
- longpulsepin,mcp

But this will create at the beginning a lot of more commands that could affect performance during the parse of the command itself (longer list of commands to check against) and the split of the command string in parts...

We can of course keep the existing command names, that maybe simplifies the whole migration.

TD-er, 2019-11-19:
> We can of course keep the existing command names, that maybe simplifies the whole migration.

Just do that and then we will find a more elegant way later.
It also makes it not a 'breaking change'.
Maybe we can come up with a much better approach once we have all these functions merged.
In my experience, you can only think of what you really want if you're using it.

Good work!

Zodiac-69, 2020-09-29:
As part of the migration, please include setting the "Port" with one command.
1. MCPPort,0xAA55 - will set all 16 output pins to the pattern - 1010 1010 0101 0101
2. MCPPort, 0xFFFF - will set all 16 pins to high(1)
3. MCPPort,0x0000 - will set all 16 pins to low(0)
4. MCPPortA,0xFF - will only set the 8 PortA pins to high(1) and not affect any PortB pins
5.MCPPortB,0xAA - will only affect the 8 PortB pins and not affect any PortA pins

giig1967g, 2020-09-29:
Hi
Nice idea. But before I am waiting for @TD-er to merge the PR.
It doesn't make sense to do changes before.