letscontrolit/ESPEasy #2755: Add support for adafruit MAX81850K breakout board

kickoo, 2019-11-19:
Is it possible to add support  Adafruit Thermocouple Amplifier with 1-Wire Breakout Board - MAX31850K

link to actual product:
https://www.adafruit.com/product/1727?fbclid=IwAR1tfROFf-7ZUa4c1-fDuZdVByx5ZBCxwQi7KPFHb7WivWIvMvcZdwIE9PU

uzi18, 2019-11-19:
do you have this module to test?

kickoo, 2019-11-19:
Yes,
I got one module.

uzi18, 2019-11-19:
what name of espeasy image do you use?

kickoo, 2019-11-19:
I'm using
Release mega-20191116

TD-er, 2019-11-19:
Hmm, I guess that adding software support is easier than operating a thermocouple amplifier in a correct way.
It is very easy to do it wrong and you probably must know quite a bit about metals.

kickoo, 2019-11-19:
Could you please explain a bit more regarding operating the thermocouple amplifier.
Have adafruit made a poor design job on this break out board?

uzi18, 2019-11-19:
@kickoo @TD-er said it is nice module and easy to use.
I will try to add support for this chip.

@kickoo please paste here sysinfo page from your espeasy.

TD-er, 2019-11-19:
Nope, I don't thing adafruit boards are bad.
The thing is that a thermocouple does work by using different metals connected to each other at the point where you want to measure the temperature.
This results in a voltage drop linear to the temperature difference.
This is a very low voltage drop, so you have to take care of noise to be able to see the difference.
And each transition from one metal to another metal (copper, tin, iron, aluminium, etc) will also act as its own thermocouple.
So there are many ways to do it wrong :)

uzi18, 2019-11-19:
thermocouple type K is very cheap, these 2 metal wires are welded together to act as thermocouple, it should be possible to do it at home ;)
but to use thermocouples as sensor is usually not so easy like with max31850k

And be sure max31850k will not accept more than 3V3 on Data line and will not work probably with "grounded" thermocouple - it will report "fault"

TD-er, 2019-11-19:
Well the chip does present itself as a 1wire sensor, so that makes it easy on our part.
As a matter of fact I think our "1wire chief engineer" is already on it, aren't you @uzi18 ? ;)

uzi18, 2019-11-19:
@TD-er yes be sure, will try to find some minutes today/tomorrow night

kickoo, 2019-11-19:
I'have been using Type K sensors for many years at work when creating thermal profiles for pcb's in reflow ovens (with bought equipment). 
This is the first time I use them at home though.

Paste info page:

Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20191116
Plugins:⋄
46 [Normal]
Build Time:⋄
Nov 16 2019 03:16:46
Binary Filename:⋄
ESP_Easy_mega-20191116_normal_ESP8266_4M1M.bin

TD-er, 2019-11-19:
Well they are very useful, but you have to make sure you are not using different metal-metal transitions which also can have different temperature.
For example copper-aluminium (just two metals) will produce a voltage difference linear to the temperature.
And the true thermocouple effect doesn't matter at all, as long as you have the same one on the opposite side and both have the same temperature. Then they will cancel each other out.

uzi18, 2019-11-19:
@kickoo I need debug log from your esp with configured Dallas task - set pin and choose module id first.
I see it is added in code but will report wrong temperature.

uzi18, 2019-11-19:
@kickoo @TD-er ok it is already supported, should work out of the box.

uzi18, 2019-11-20:
what we can add is use NAN if sensor fault is setted

Misiu, 2019-11-20:
I have MAX31865 board ([this one](https://www.aliexpress.com/item/32916449521.html)) so I can help with testing.
@uzi18 I think sensor fault should be configurable as with DS18B20 (I only used this sensor, sorry). The user has the ability to specify what to report on fault: -127, NaN or other value

uzi18, 2019-11-20:
@Misiu but MAX31865 has SPI interface not 1-wire, yes I think about use already defined option :)

uzi18, 2019-11-20:
@kickoo still waiting for debug log from your module connected to espeasy

kickoo, 2019-11-20:
Sorry for late reply, I'm working late.
just for clarification (for me).
With bug log you mean  Web Log Level : Debug dev?
[log max31850.txt](https://github.com/letscontrolit/ESPEasy/files/3870408/log.max31850.txt)

uzi18, 2019-11-20:
@kickoo ok so it works perfect :)

```
932615: DS: SP: 98,1,f0,16,f0,ff,ff,ff,54,OK,P,8
932617: DS   : Temperature: 25.50 (3b-4c-96-5d-6-d8-6c-fd)
942733: DS: SP: 98,1,f0,16,f0,ff,ff,ff,54,OK,P,8
942735: DS   : Temperature: 25.50 (3b-4c-96-5d-6-d8-6c-fd)
952849: DS: SP: 98,1,f0,16,f0,ff,ff,ff,54,OK,P,8
952851: DS   : Temperature: 25.50 (3b-4c-96-5d-6-d8-6c-fd)

0x0198*0,0625 = 25,5deg.
```

kickoo, 2019-11-20:
Yes it seems to be working. 
I have no idea about the accuracy though.
It seems to be a bit off but then  I got a chinese  sensor.

uzi18, 2019-11-20:
error should be not more than +/-1deg. (for <100deg.) to +/- 4 or 6deg.  (for >700deg.)

kickoo, 2019-11-20:
It seems like my hunch was wrong.
I hooked up an DS18b20 for reference and an ordinary thermometer. All of them was within 1,1 deg.

What does error state value affect?

uzi18, 2019-11-20:
check properly connections first, we know about some fake ds18b20, 
please give put here debug log from ds18b20

kickoo, 2019-11-20:
[log max31850_1.txt](https://github.com/letscontrolit/ESPEasy/files/3871198/log.max31850_1.txt)

uzi18, 2019-11-20:
@kickoo they work ok - about should show 22 deg. so what is your problem?

kickoo, 2019-11-20:
Nothing is wrong, 
I admitted  earlier I was wrong about that the type K sensorwas off by some degrees.

I thought It felt like 18 degrees  but obviously was wrong when three different thermometers showed  about 22 deg.  I need to put on a sweater :-)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er