letscontrolit/ESPEasy #2758: Add a choice for IP/Hostname of controller in Custom.h

chemmex, 2019-11-20:
So far, only IP address of controller can be defined as DEFAULT_SERVER in the Custom.h
To keep consistency with web interface, there should be a choice of ip/hostname of the server, as per https://github.com/letscontrolit/ESPEasy/issues/2408

TD-er, 2019-11-22:
Can you have a look at PR #2764 to see if this is what you meant?

chemmex, 2019-11-22:
Yes, exactly. It compiles, but does not enable the controller

TD-er, 2019-11-22:
> Yes, exactly. It compiles, but does not enable the controller

Not?

This line suggests it should enable it if you set it in the Custom.h file:
https://github.com/letscontrolit/ESPEasy/blob/948039dcdaba54eda73651ce61214936e8220f16/src/Misc.ino#L1192

chemmex, 2019-11-22:
Great, thanks. Closing the issue.

PR #2764 (2019-11-22): [Defaults] Add Hostname option for default controller (#2758)
Fixes: #2758
commit 948039dc (2019-11-22): [Defaults] Add Hostname option for default controller (#2758)

Fixes: #2758
closed 2019-11-22 by chemmex