letscontrolit/ESPEasy #2759: event call order

giig1967g, 2019-11-21:
Hi @TD-er ,
in my original development of the `monitor `command I had set the call of PLUGIN_MONITOR before the call to the PLUGIN_TEN_PER_SECOND.

I noticed that you have swapped the order (inside ESPEasy.ino) and now the call to the PLUGIN_MONITOR is after PLUGIN_TEN_PER_SECOND.

Because of the way it has been designed, the PLUGIN_MONITOR is expecting to run before the PLUGIN_TEN_PER_SECOND event, or some unexpected situations may occur (this is why some events were not generated). 

Is there any particular reason why you have swapped the two events?
If there are good reasons, I can modify the PLUGIN_MONITOR to expect to be executed after the PLUGIN_TEN_PER_SECOND, or I can invert the two events as they were before.

TD-er, 2019-11-21:
I don't know why that order was changed.

giig1967g, 2019-11-21:
I will switch it back and add a comment.

closed 2019-11-21 by giig1967g