letscontrolit/ESPEasy #2760: P036 lines cannot have more than 32 chars, P023 can have longer lines. Multiple values cannot be shown due to this limitation.

Roos-AID, 2019-11-21:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [X] filled out all applicable fields below.

### Steps already tried...
- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

With the P036 Display - OLED SSD1306/SH1106 Framed it is not possible to enter more than 32 chars in Line 1 to 8. 
P023 Display - OLED SSD1306 the lines can be much longer. 
Due to this restriction it is not possible to have multiple values in a single line on P036  for example :
"[Klima2#CO2_PPM]PPM - [Klima2#CO2_Temp]C{D}" does not fit. 

With P023 this is not a problem. 

### Expected behavior

It is expected that multiple values with text can be entered in Lines 1 to 8, for example 
on Line 1 : Hum "[Klima1#Humidity]% [Klima1#Temperature]C{D}" and Line 2 : 
"[Klima2#CO2_PPM]PPM - [Klima2#CO2_Temp]C{D}"

### Actual behavior

The lines are limited to 32 chars :  eg "[Klima1#Humidity]% [Klima1#Tempe"

### Steps to reproduce

1. Add device P036 and try to enter longer text lines

Yes

### System configuration

Hardware:

Build:⋄ | 20104 - Mega
System Libraries:⋄ | ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191108

ESP Board Name: | PLATFORMIO_ESP12E
I2C scan : address 0x3C  PCF8574A OLED

Software (see version tried below).
ESP_Easy_mega-20191116_normal_ESP8266_4M1M.bin

ESP Easy version: 

Tried builds 
ESPEasy_mega-20191116
ESPEasy_mega-20191113
ESPEasy_mega-20191108

For now this is not high priority as fall-back solution is to temporarily use P023

PS. everybody is doing a great job with this device, I keep finding more possibilities every day !

Regards,
Rob

uzi18, 2019-11-21:
@uwekaditz maybe you have answere?

TD-er, 2019-11-21:
If we increase the number of characters on a line, we may break existing settings for this plugin.

uwekaditz, 2019-11-21:
I can try to add an additional array of 32 characters per line (in sum 12 lines and total 384 bytes) to the **_CustomTaskSettings_** but keeping the existing settings for this plugin.
I've already thought about adding additional data (e.g. an array of line settings) to the CustomTaskSettings of this plug in order to create individual font settings for each line.
The size of the CustomTaskSettings is limited to 1024 bytes per plug in, so there is still space to play with.

Roos-AID, 2019-11-21:
Isn't it possible to deprecate P036 in future release and release a new module with all features ? Or document that with new release of P036 the Device settings must be manually reviewed ?
Personally I always test each new release on all devices before deploying it in production in my Home automation.

uwekaditz, 2019-11-21:
So far, the P036 plugin has always worked with 32 characters per line. I didn't change this setting!

Roos-AID, 2019-11-21:
Indeed, it is  a feature request , not a bug.  You did a great job with all the new features, that made we try to add more into the line than possible now.

uwekaditz, 2019-11-22:
I created a new pull request for [P036]
[https://github.com/letscontrolit/ESPEasy/pull/2763]
**CHG:** Each line can now have up to 64 chars (old settings with 32 chars are kept!)
**FIX:** Overlapping while page scrolling (size of line content for scrolling pages is now limited to 128 pixel)

PR #3050 (2020-07-23): [P036] Memory optimization (issue #2799) and other improvements
CHG: Memory optimization (issue #2799)
CHG: Added setting for 'Disable scrolling while WiFi is not connected',
         scrolling is not smooth as long as the ESP tries to connect Wifi (Load 100%!)
BUG: Wifi-Bars not fully right for 128 pixel wide displays
BUG: clearing window for scrolling lines was too high and to narrow
BUG: content of lines with pixel width > 255 had to be shortened
CHG: scrolling lines improved
CHG: variables renamed for better understanding

Fixes: #2312
Closes: #3026 
Fixes: #3127 
Fixes: #2760 
Fixes: #279 [... cut]
closed 2020-07-23 by TD-er