letscontrolit/ESPEasy #2762: Different behaviours on events

giig1967g, 2019-11-21:
Hi all,
During the troubleshooting of the GPIO plugins, I noticed different behaviour in the events generated by the P001, P009 and P019 plugin.

Following a `gpio/pcfgpio/mcpgpio` command:
For P001, [task#value] event is generated **only** if the state of the GPIO changes
For P009 and P019, [task#value] event is generated **always**, even if the state of the GPIO doesn't change.

For P001, P009, P019 monitor event is generated **always** even if the state of the GPIO doesn't change.

So, what should be the expected events, in order to fix it?

TD-er, 2019-11-21:
Event on change, not on sending a command to set a value.

At least that makes the most sense to me.

tonhuisman, 2023-09-02:
This seems no longer relevant, and can be closed.

closed 2023-09-02 by TD-er