letscontrolit/ESPEasy #2770: WiFi Connected time overflows after 49.7 days

TD-er, 2019-11-22:
These are the more serious issues.

Unit Number: | 7
-- | --
Local Time: | 2019-11-23 00:30:53
Uptime: | 60 days 0 hours 28 minutes
Load: | 8.00% (LC=344)
CPU Eco Mode: | true
Boot: | Cold boot (0)
Reset Reason: | Software/System restart
Wifi: | 802.11G (RSSI -61 dB)
Connected: | 10d07h25m
Last Disconnect Reason: | (1) Unspecified
Number Reconnects: | 0

As can be seen, the connected time is overflowing.
Maybe the time to connect should be stored in UTC and not in `millis()`.

thomastech, 2019-11-23:
Good catch! That's not a problem that us mortals would experience. :)

Internally saving uptime in UTC format sounds like a reasonable workaround. Or continue to use the millis() counter and reset it every 24 hours. An int16 could keep track of the day++ count.

TD-er, 2019-11-23:
>  An int16 could keep track of the day++ count.

Better use an uint16?

thomastech, 2019-11-23:
> Better use an uint16?

Yes, uint16 is better. The int16 would wrap after 93 years, which might be a problem for some users. :)

- Thomas

TD-er, 2019-11-23:
Yep and quite hard to reproduce ;)

Heinz-Peter, 2020-01-24:
702 days in real :-) have a look .... regards Heinz Peter
[image]

TD-er, 2020-01-24:
Wow! :)

Grovkillen, 2020-01-25:
Wow!!

thomastech, 2020-01-25:
My Mega branch loaded board demands an apology from your V2.0.0-dev12 board. We're living in the #MeToo era and your board's uptime bragging is nothing more than cruel byte shaming.

Don't tell my board I said it, but that is indeed an impressive uptime. :)

- Thomas

tonhuisman, 2023-09-02:
This is no longer relevant, so can be closed.

closed 2023-09-02 by TD-er