letscontrolit/ESPEasy #2771: LongPulse in rules affects WiFi LED behaviour in 20191122_test_ESP8266

mrdc, 2019-11-23:
Hello,

After updating to ESP_Easy_mega-20191122_test_ESP8266_4M_VCC I've noticed that LongPulse in rules affects WiFi LED behaviour:
WiFi LED is turned off in Settings, but LongPulse activated via Rules turnes the WiFi LED ON then after few seconds OFF and then again ON.
[image]

[image]

My Rules:
[image]

TD-er, 2019-11-23:
Is this new behavior in this build?
Is the unit configured to be working in deep sleep? (GPIO 16 connected to RST)

mrdc, 2019-11-23:
Yes, no changes were made to ESPEasy config - I've updated via OTA from `ESPEasy_mega-20191003`
Builds `ESPEasy_mega-20190630, ESPEasy_mega-20190903, ESPEasy_mega-20191003` worked well with the config.

> Is the unit configured to be working in deep sleep? (GPIO 16 connected to RST)

No.

[image]

TD-er, 2019-11-23:
Was it working in a version before 20191114?
The last change related to longpulse was this one: https://github.com/letscontrolit/ESPEasy/commit/ab7cc2decc0157893050d5eb08d298a1539365be#diff-06582a14b89c33aaeb4c404ff562c4ba

mrdc, 2019-11-23:
I was using 20191003_test before 20191122_test. Haven't checked between 20191003 and 20191122.

giig1967g, 2019-11-23:
i can't reproduce it with your rules.
can you send the screenshot of the devices?

mrdc, 2019-11-23:
So, I've found what is wrong: first it's not WiFi LED but Power LED - I'm using NodeMCE v2 and it has Power LED on GPIO16, somehow LongPulse (I used it for GPIO16-D0 for dust sensor) was not working properly BEFORE 20191122_test build, so that's why it wasn't blinking with my Rules. Now LongPulse works and blinks according to my Rules.
[image]

TD-er, 2019-11-23:
So we must conclude this issue a confirmation of a proper fix of another issue?

mrdc, 2019-11-23:
Yes, looks like LongPulse was fixed after build `20191003 `.

closed 2019-11-23 by mrdc