letscontrolit/ESPEasy #2772: Problems with Nextion

schamu, 2019-11-23:
I have an error with the Nextion plugin:
 
The command does not work in the rules:
nextion, page99.t13.txt = "% c_m2dhm% (% uptime%)"
 
The command works:
nextion, page99.t14.txt = "% ip%"
 
But it works in lines 1-10 for devices.
 
The following is displayed in the logfile:
1311958: EVENT: Rules # Timer = 7
1312012: ACT: nextion, page99.t13.txt = '0d00h22m'
1312020: ACT: nextion, page99.t14.txt = '192.168.178.14'

The following are also not sent to the Nextion:
Nextion, page01.t10.txt = "[BME280 # Temperature]"
Nextion, page01.t11.txt = "[# BME280 Humidity]"
Nextion, page01.t12.txt = "[BME280 # Pressure]"
Nextion, page05.t13.txt = "[DS18b20 # Temperature]"

When I send UPD commandos to the Nextion it works too.
Only the rules do not work.

The error occurs at least since version mega-20191003

TD-er, 2019-11-23:
See this issue describing what you probably experience: https://github.com/letscontrolit/ESPEasy/issues/2724

For example:
```
nextion, page99.t13.txt="% c_m2dhm%(% uptime%)"
```
Will then become:
```
nextion,'page99.t13.txt="% c_m2dhm%(% uptime%)"'
```
(added single quote to wrap the parameter)

schamu, 2019-11-23:
There is no space in my rules, it was a mistake in the post. I've already tried that with the quotes.

TD-er, 2019-11-23:
The quotes wrapping the whole parameter?

schamu, 2019-11-23:
yes

TD-er, 2019-11-23:
Please post your relevant rules, which you have now active so I can see what is stored in your rules.
Wrap them in 3x left quote (`) to make sure the syntax doesn't get mangled here in a comment.

schamu, 2019-11-23:
`on Rules#Timer=7 do
 //Nextion,page99.t11.txt="0%sunrise% - %sunset%"
 nextion,'page99.t13.txt="%c_m2dhm%(%uptime%)"'
 nextion,page99.t14.txt="%ip%"
endon

on Rules#Timer=8 do
 Nextion,page01.t10.txt="[BME280#Temperature]"
 Nextion,page01.t11.txt="[BME280#Humidity]"
 Nextion,page01.t12.txt="[BME280#Pressure]"
 Nextion,page05.t13.txt="[DS18b20#Temperature]"
endon

on SendeSchalter do
 Nextion,page06.bt0.val=[schalter#sofa]
 Nextion,page06.bt1.val=[schalter#lichterkette]
 Nextion,page06.bt2.val=[schalter#terasse]
 Nextion,page06.bt3.val=[schalter#steckdose]
endon`

TD-er, 2019-11-23:
Can you enable debug log and post the relevant log parts where the decoding of the rules are shown?

schamu, 2019-11-23:
`3723967: Command: event
3723967: EVENT: SendeSchalter
3724084: ACT : Nextion,page06.bt0.val=1
3724115: ACT : Nextion,page06.bt1.val=0
3724144: ACT : Nextion,page06.bt2.val=0
3724174: ACT : Nextion,page06.bt3.val=0
3728911: EVENT: Rules#Timer=7
3728962: ACT : Nextion,page99.t13.txt=''0d01h02m''
3728967: ACT : Nextion,page99.t14.txt='192.168.178.14'
3732918: EVENT: Rules#Timer=8
3733004: ACT : Nextion,page01.t10.txt='20.7'
3733016: ACT : Nextion,page01.t11.txt='50.9'
3733029: ACT : Nextion,page01.t12.txt='984.6'
3733040: ACT : Nextion,page05.t13.txt='20.7'`

TD-er, 2019-11-23:
Is that with debug level enabled?

schamu, 2019-11-23:
were can I enable the debug level?

TD-er, 2019-11-23:
Tools => Advanced
[image]

schamu, 2019-11-23:
4448920: EVENT: Rules#Timer=7
4449003: ACT : Nextion,page99.t13.txt=''0d01h14m''
4449007: ACT : Nextion,page99.t14.txt='192.168.178.14'
4450988: SW : GPIO=12 State=0 Output value=0
4450993: EVENT: taste4#1=0.00
4451039: ACT : Publish /ESP_Display1/Bewegung,0
4451054: Command: publish
4451055: Publish: /ESP_Display1/Bewegung0
4452914: EVENT: Rules#Timer=8
4453005: ACT : Nextion,page01.t10.txt='20.9'
4453017: ACT : Nextion,page01.t11.txt='50.5'
4453029: ACT : Nextion,page01.t12.txt='984.2'
4453040: ACT : Nextion,page05.t13.txt='20.8'

TD-er, 2019-11-23:
What build are you using?
If that's with the Debug level set, then we're missing the actual debug log entries.
It looks like `BUILD_NO_DEBUG` is defined, but that should only be for "minimal" or "normal" builds (or custom builds you made yourself)

schamu, 2019-11-23:
Build:⋄ | 20104 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2_6_1, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191123
Plugins:⋄ | 80 [Normal] [Testing] [Development]
Build Md5: | 835d9c4739177bbad2821062ba8a35
Md5 check: | passed.
Build Time:⋄ | Nov 23 2019 03:12:49
Binary Filename:⋄ | ESP_Easy_mega-20191123_dev_ESP8266_4M1M.bin

TD-er, 2019-11-23:
I'm looking at the code of the Nextion plugin now.
Apparently the command for the Nextion display is actually the name of the plugin.
Is that true?

Also the Nextion plugin does not use the (new) standard parse functions, although I am not 100% sure that's what is causing the issue here.

schamu, 2019-11-23:
In this build is all ok;

Build:⋄ | 20103 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20190827
Plugins:⋄ | 81 [Normal] [Testing] [Development]
Build Md5: | 6dd455f8fbf8143bdd11bd953a4b2f
Md5 check: | passed.
Build Time:⋄ | Aug 27 2019 02:34:26
Binary Filename:⋄ | ESP_Easy_mega-20190827_dev_ESP8266_4M.bin

TD-er, 2019-11-23:
Can you test with [this test build](https://www.dropbox.com/s/v79uprlomajqof0/dev_ESP8266_4M1M_Nextion_2772.bin?dl=0) ?

schamu, 2019-11-23:
This one?  There ist the same error

Build:⋄ | 20104 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2.7.0-dev stage, NONOS SDK 2.2.2-dev(bb83b9b), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191123
Plugins:⋄ | 79 [Normal] [Testing]
Build Md5: | a667330ae76d2cfa961f72db502680
Md5 check: | passed.
Build Time:⋄ | Nov 23 2019 03:49:28
Binary Filename:⋄ | ESP_Easy_mega-20191123_test_beta_ESP8266_4M1M.bin

schamu, 2019-11-23:
Sorry, now the right one.

Sieht gut aus, ich werde noch alles durchtesten

schamu, 2019-11-23:
It looks like everything works.
Thank you for your good work

TD-er, 2019-11-23:
OK, I will make it into a pull request and if someone else can test it, then it will be in the next nightly build.

thomastech, 2019-11-24:
> ... and if someone else can test it, then it will be in the next nightly build.

Done! I confirmed that your test build restores the broken Nextion rules.

- Thomas

TD-er, 2019-11-24:
While you're at it... I just finished changing other plugins too, which were probably effected in the same way.
Test build is started, but if you don't mind, I will now try to get some sleep ;)

TD-er, 2019-11-24:
The test build for this PR:
[ESPEasy_mega-20191108-36-PR_2728.zip](https://www.dropbox.com/s/g2xk0884j02dx0d/ESPEasy_mega-20191108-36-PR_2728.zip?dl=0)

schamu, 2019-11-24:
I tested this version on different controllers. I use the 8266_4M1M dev and normal versions with different sensors and switches / buttons and mqtt.

All functions work for me.

Thank you for your great work.

PR #2774 (2019-11-24): [Nextion] Fix parsing rules for Nextion plugin (#2772)
Fixes: #2772
commit c63f3a28 (2019-11-24): [Nextion] Fix parsing rules for Nextion plugin (#2772)

Fixes: #2772
closed 2019-11-24 by TD-er