letscontrolit/ESPEasy #2773: P2P and Dummy Device automatic update impossible to transfert data to variable when not enabled

PPatou, 2019-11-23:
Hello,
Since version ESP_Easy_mega-20191116_normal_ESP8266_4M1M.bin there is a problem with the transfer of data with P2P protocol.
I explain. 
In receiving unit the task is not created, in the sending unit the task is created and submitted.
Then in the receiving unit the task is automatically created and values can be seen in the device tap. At that moment the task is disabled.
With a disabled task the values cannot be transfer to an internal variable for example with the command ` let,1,[Data11#Temp]  `
If I enabled the task 11 then the value disappears in the device tab and I have to delete the task in the receiving unit and resubmit the task in the sending unit.
Can you please investigate ?
Thanks in advance
Patou

TD-er, 2019-11-23:
Is the plugin you try to send present in the build running on the receiving node?

For example you want to send "GPS" data from node 1 to node 2, but node 2 does not have the GPS plugin included in the build.

PPatou, 2019-11-23:
I transfers only from Dummy devices so this should be included in every build !

tonhuisman, 2023-09-02:
This seems to have been either resolved, or no longer relevant, so can be closed.

closed 2023-09-02 by TD-er