letscontrolit/ESPEasy #2775: PLUGIN_READ triggers every second Plugin_053

red55der, 2019-11-24:
I have pulled Release mega-20191123 to test the Plugin_053 and tweak it for my situation.
I added a log line in the PLUGIN_READ and see that it's triggered every second.
The interval in the Device settings is set on 30 seconds.

The PLUGIN_READ should be called every x moment where x = the setted interval in the Device setting.

PLUGIN_READ is called every second.

1. Add a logline to Plugin_053 --> PLUGIN_READ 

Yes

Hardware:
Wemos D1 4M, PMS7003.

ESP Easy version: Release mega-20191123

uzi18, 2019-11-24:
what did you changed?

TD-er, 2019-11-24:
The code of that plugin is:
```c++
    case PLUGIN_READ:
      {
        // When new data is available, return true
        success = values_received;
        values_received = false;
        break;
      }
```
So the data is being read in the 10/sec call and the PLUGIN_READ does only return success=true, when something is read.

By the way, you can also see in the timing stats how often it is being called.
The PLUGIN_READ should be called every N seconds, which is what you have set as interval in the task configuration.

Some plugins (like the BME280) can re-schedule themselves to perform a READ, but I don't see this in the P053 code.

mackowiakp, 2019-11-26:
Because this thread has the "Needs Info" label, I wanted to post my thoughts here. I note that these are "observations" and nothing else. Just my "5 cents".
Below screenshots from one of my ESP units. The configuration is as follows:
- ADC
- Switch (digital output)
- PMS dust sensor
- DTH22
- BMP180
- Generic - uptime

As You can see, all plugins call all PLUGIN_PER_<SOMETHING> together with PLUGIN_READ and others. And this even when no PLUGIN_PER_<SOMETHING> is defined in the source code for the given sensor.
To be honest, I don't quite understand it, but it shows "Timing stats"
Screenshot below:

[image]

[image]

[image]

TD-er, 2019-11-26:
> As You can see, all plugins call all PLUGIN_PER_SOMETHING together with PLUGIN_READ and others. And this even when no PLUGIN_PER_SOMETHING is defined in the source code for the given sensor.

That's because the statistics are collected where the call is being made.
When not implemented in the plugin, it returns success = false.
But there are also valid use cases where an implemented call does return this false situation.
So at the place where the statistics are collected, you don't know whether it is implemented or not.
Also these are statistics for me as a developer, not really for day-to-day use, so it doesn't need to be handling all special cases.

Statistics on "not implemented" situations can also be useful by the way.

tonhuisman, 2022-05-07:
This doesn't seem relative anymore?

closed 2022-05-07 by TD-er