letscontrolit/ESPEasy #2777: Device Task Values are not published to MQTT

andig, 2019-11-24:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
```
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 2_6_1, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20191123
Plugins:⋄
46 [Normal]
Build Time:⋄
Nov 23 2019 03:34:36
Binary Filename:⋄
ESP_Easy_mega-20191123_normal_ESP8266_4M1M.bin
```
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I have flashed a WeMOS D1 with latest firmware (see above). Added two 1wire DS18b20 sensors:

[image]

Setup MQTT controller:

[image]

Ensures MQTT is connected:

```
❯ mosquitto_sub -h nas.fritz.box -p 1883 -t '#' -v
ESP_Easy/status/LWT moin
```

### Expected behavior

Expect 1wire measurements to appear under the MQTT topic:

[image]

### Actual behavior

Only the LWT is shown but no values for the active tasks are published.

### Steps to reproduce

Not sure- the entire setup is described above.

Always.

### System configuration

Hardware: Wemos D1 Mini, 2 1wires.

ESP Easy version:  ESP_Easy_mega-20191123_normal_ESP8266_4M1M.bin

### Rules or log data

```
590006 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
591213 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
591236 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
592443 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
592466 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
593674 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
593696 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
594904 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
594927 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
596135 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
596158 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
597365 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
597388 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
598596 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
598619 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
599826 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
599849 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
601057 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
601080 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
602287 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
602310 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
602377 : Info  : WD   : Uptime 10 ConnectFailures 236 FreeMem 19152 WiFiStatus 3
603517 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
603540 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
604747 : Info  : DS   : Temperature: 21.50 (28-aa-36-d1-1d-13-2-44)
604770 : Info  : DS   : Temperature: 21.00 (28-b5-d1-4d-33-14-1-af)
```

TD-er, 2019-11-24:
What MQTT controller are you using?
Can you subscribe to the topics using a MQTT viewer like MQTT box (or something else), so we can see what's going on?
Maybe also look at the log with level set to debug, to see if that does give some more info.

andig, 2019-11-24:
> What MQTT controller are you using?

Homeassisstant/OpenHAB

> Can you subscribe to the topics using a MQTT viewer like MQTT box (or something else), so we can see what's going on?

See `mosquitto_sub` output above- nothing else is going on.

> Maybe also look at the log with level set to debug, to see if that does give some more info.

There is no MQTT activity at dev log level:

```
2462621 : Debug : DS: SP: 50,1,4b,46,1f,ff,1f,10,c1,OK,7
2462622 : Info  : DS   : Temperature: 21.00 (28-aa-36-d1-1d-13-2-44)
2463820 : Debug : DS: SP: 50,1,4b,46,1f,ff,1f,10,c1,OK,7
2463821 : Info  : DS   : Temperature: 21.00 (28-aa-36-d1-1d-13-2-44)
2463852 : Debug : DS: SP: 48,1,4b,46,1f,ff,1f,10,bf,OK,7
2463852 : Info  : DS   : Temperature: 20.50 (28-b5-d1-4d-33-14-1-af)
```

andig, 2019-11-24:
Arrghh... nI need to check "Send to controller" in the task- now its working. Sorry!!!

closed 2019-11-24 by andig