letscontrolit/ESPEasy #2780: feature request - lightweight version for OTA

risros, 2019-11-24:
Since OTA upgrade alows firmware sizes of up to ~600kB, which makes upgrade to mega release impossible, why not making a tiny release, with a minimal memory footprint, which would be an intermediate step on every OTA upgrade.
The size of this firmware should be such, that mega release would be possible to upgrade to over OTA.
All settings must be preserved.

TD-er, 2019-11-24:
You mean the "minimal" builds we have included in the nightly builds?

risros, 2019-11-24:
Yes. But with the only one purpose. To be able to upgrade to larger builds.

TD-er, 2019-11-24:
That one is included in the ZIP file for the nightly builds

risros, 2019-12-08:
@TD-er , I'm trying to update using OTA and 2-step build.
But when pressing "Update Firmware" button, I get a page /update with content:
```
URI: /update
Method: GET
Arguments: 0
```
Is OTA enabled at all? Because the description next to the button is confusing:

Load a new firmware WARNING OTA not possible. Max sketch size: 599 kB (614384 bytes)

Is it possible or not?
Thanks

TD-er, 2019-12-08:
What build version are you running?
What size is claimed to be free for the sketch?

risros, 2019-12-08:
20191208. But same on older releases

TD-er, 2019-12-08:
Please give a bit more info on the amount of **free** sketch space.
What sketch filename is currently loaded?
It sounds like you have not flashed the "minimal OTA" version.

You can just cut and paste the sysinfo page here and when pasting, the info will be formatted again in table layout.

risros, 2019-12-09:
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20191108
Plugins:⋄
46 [Normal]
Build Time:⋄
Nov  8 2019 03:13:16
Binary Filename:⋄
ESP_Easy_mega-20191108_normal_ESP8285_1M.bin

TD-er, 2019-12-09:
label | value
--- | ---
Sketch Size: | 848 kB (24 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | false
OTA 2-step Needed: | true

As you can see, the sketch you have uploaded to the ESP is larger than the max. OTA sketch size for that flash size.
So OTA is disabled since you would not get a working node anyway when trying to perform OTA.

The smallest 2-step OTA image we have is roughly 280 kB, so on an 1M node you have 128k SPIFFS, 280k smallest OTA image => 599 k max. sketch free if you plan to do 2-step OTA.
This means you need to load a sketch smaller than 599 k if you ever plan to do OTA.

Let's assume you had loaded a sketch of 700k, then the 2-step OTA image would fit on the 1M flash, but by performing the OTA, you would destroy your SPIFFS filesystem. So that's why I put the limit at 599k for the 1M nodes.

So to upgrade this one, you need to do it via serial. Sorry.

risros, 2019-12-13:
Thank you. I understand now.
Perhaps the text next to the Firmware Button should read "Current image size exceeded..."

closed 2019-12-13 by risros