letscontrolit/ESPEasy #2782: 'StaticJsonDocument' was not declared in this scope

s0170071, 2019-11-25:
As soon as I add #define P016_P035_Extended_AC to custom.h, I get a compile error: 

'StaticJsonDocument' was not declared in this scope

my Custom.h:

```
#ifndef ESPEASY_CUSTOM_H
#define ESPEASY_CUSTOM_H

/*
    To modify the stock configuration without changing the EspEasy.ino file :

    1) rename this file to "Custom.h" (It is ignored by Git)
    2) define your own settings below
    3) define USE_CUSTOM_H as a build flags. ie : export PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'"
 */

/*

 #######################################################################################################
   Your Own Default Settings
 #######################################################################################################

    You can basically ovveride ALL macro defined in ESPEasy.ino.
    Don't forget to first #undef each existing #define that you add below.
    But since this Custom.h is included before other defines are made, you don't have to undef a lot of defines.

    Here are some examples:
 */

#ifdef BUILD_GIT
# undef BUILD_GIT
#endif // ifdef BUILD_GIT

#define BUILD_GIT           "My Build: "  __DATE__ " "  __TIME__

#define DEFAULT_NAME        "MyEspEasyDevice"                        // Enter your device friendly name
#define UNIT                0                                        // Unit Number
#define DEFAULT_DELAY       60                                       // Sleep Delay in seconds

// --- Wifi AP Mode (when your Wifi Network is not reachable) ----------------------------------------
#define DEFAULT_AP_IP       192, 168, 1, 1                           // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_SUBNET   255, 255, 255, 0                         // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_KEY      "configesp"                              // Enter network WPA key for AP (config) mode

// --- Wifi Client Mode -----------------------------------------------------------------------------
#define DEFAULT_SSID                         ""            // Enter your network SSID
#define DEFAULT_KEY                          "" // Enter your network WPA key
#define DEFAULT_USE_STATIC_IP                false                   // (true|false) enabled or disabled static IP
#define DEFAULT_IP                           "192.168.0.50"          // Enter your IP address
#define DEFAULT_DNS                          "192.168.0.1"           // Enter your DNS
#define DEFAULT_GW                           "192.168.0.1"           // Enter your Gateway
#define DEFAULT_SUBNET                       "255.255.255.0"         // Enter your Subnet
#define DEFAULT_IPRANGE_LOW                  "0.0.0.0"               // Allowed IP range to access webserver
#define DEFAULT_IPRANGE_HIGH                 "255.255.255.255"       // Allowed IP range to access webserver
#define DEFAULT_IP_BLOCK_LEVEL               1                       // 0: ALL_ALLOWED  1: LOCAL_SUBNET_ALLOWED  2:
// ONLY_IP_RANGE_ALLOWED
#define DEFAULT_ADMIN_USERNAME               "admin"

#define DEFAULT_WIFI_CONNECTION_TIMEOUT      10000 // minimum timeout in ms for WiFi to be connected.
#define DEFAULT_WIFI_FORCE_BG_MODE           false // when set, only allow to connect in 802.11B or G mode (not N)
#define DEFAULT_WIFI_RESTART_WIFI_CONN_LOST  false // Perform wifi off and on when connection was lost.
#define DEFAULT_ECO_MODE                     false // When set, make idle calls between executing tasks.
#define DEFAULT_WIFI_NONE_SLEEP              false // When set, the wifi will be set to no longer sleep (more power
// used and need reboot to reset mode)
#define DEFAULT_GRATUITOUS_ARP               false // When set, the node will send periodical gratuitous ARP
// packets to announce itself.

// --- Default Controller ------------------------------------------------------------------------------
[... 213 lines cut ...]
// #define USES_C007   // Emoncms
// #define USES_C008   // Generic HTTP
// #define USES_C009   // FHEM HTTP
// #define USES_C010   // Generic UDP
   #define USES_C011   // Generic HTTP Advanced
// #define USES_C012   // Blynk HTTP
   #define USES_C013   // ESPEasy P2P network
   #define USES_C014   // homie 3 & 4dev MQTT
// #define USES_C015   // Blynk
// #define USES_C016   // Cache controller
// #define USES_C017   // Zabbix
// #define USES_C018   // TTN/RN2483

/*
 #######################################################################################################
 ###########     Notifiers
 #######################################################################################################
 */

// #define USES_N001   // Email
// #define USES_N002   // Buzzer

#endif // ESPEASY_CUSTOM_H

```

s0170071, 2019-11-25:
resolved it by adding
`#include "ArduinoJson.h"`
to P035_IRTX.ino
Still, this should not happen...

TD-er, 2019-11-25:
I think this include is also needed for `_P016_IR.ino` as the same class is used in there.

TD-er, 2019-11-25:
Can you make a PR for the addition of both includes?

s0170071, 2019-11-25:
hmm. Now it compiles but gives me an error in the log. 
Deserialize Json failed: NoMemory

Other than that, I am having a hard time copying that line from the weblog as it jumps away then I try to do so. Yes, I have unchecked Autoscroll. Try strg+f "failed". You see it but you cannot grab it :-o

TD-er, 2019-11-25:
There is a copy button on the web page. Then you can copy/paste it into notepad or something similar.

jimmys01, 2019-11-26:
> Deserialize Json failed: NoMemory

what is the command you are giving it? I suppose is an irsendAC command

In line 210 of P035.ino   `StaticJsonDocument<300> doc;`
Make it bigger  `StaticJsonDocument<320> doc;`
or even 350 if this doesnt work for you.
But i am curious how big is your command

TD-er, 2019-11-26:
Is it possible there to include the needed size in the log?

jimmys01, 2019-11-26:
I dont think so

TD-er, 2019-11-26:
Well if you know how much the message is you want to encode, you can give a rough estimate since you know the JSON structure and thus have a fixed offset.

jimmys01, 2019-11-26:
well I have done this and I had concluded that 300 would be enought

s0170071, 2019-11-27:
The command that I send is: 
`
IRSENDAC,{"power":"OFF","temp":"16","protocol":"FUJITSU_AC","model":"ARREB1E","mode":"AUTO", "swingv":"OFF", "swingh":"OFF", "turbo":"OFF", "econo":"OFF", "fanspeed":"AUTO"}
`
Thats 174 characters. Sounds like it should have worked. In fact, the ESPEasy log error is gone although I did not touch anything since.

The log shows now: 
```
110256038: IRTX: JSON received: {'power':'OFF','temp':'16','protocol':'FUJITSU_AC','model':'ARREB1E','mode':'AUTO', 'swingv':'OFF', 'swing
110256115: IRTX: IR Code Sent: {'power':'OFF'
```

Btw, this does compile of of the box: 
```
//StaticJsonDocument<300> doc;
        DynamicJsonDocument doc(string.length());

```

Cant't test it right now, its too early in the morning to wake everyone by playing with the AC. Will report back.

jimmys01, 2019-11-27:
Encapsulate the whole JSON inbetween '

 `IRSENDAC,' {"power":"OFF","temp":"16","protocol":"FUJITSU_AC","model":"ARREB1E","mode":"AUTO", "swingv":"OFF", "swingh":"OFF", "turbo":"OFF", "econo":"OFF", "fanspeed":"AUTO"}'`

TD-er, 2019-11-27:
> Cant't test it right now, its too early in the morning to wake everyone by playing with the AC. Will report back.

You don't have a 'testing' and 'production' environment for that? ;)

TD-er, 2019-11-27:
> Encapsulate the whole JSON inbetween '
> 

Yep, see also https://github.com/letscontrolit/ESPEasy/issues/2724

s0170071, 2019-11-27:
> You don't have a 'testing' and 'production' environment for that? ;)

I wish I had. Whenever I do a 
`git clone MyHouse; rm -s /tmp/wife`
I get an access denied error 👎  

> Encapsulate the whole JSON inbetween '

I'll try but I am pretty sure it worked without. The command comes from OpenHAB via MQTT and I did not touch that part of my setup. I just updated the ESP.

s0170071, 2019-11-27:
> recent changes regarding more strict parsing of parameters.

Ah, I get it.

jimmys01, 2019-11-27:
Check also https://www.letscontrolit.com/forum/viewtopic.php?f=2&t=6850
I have updated the posts to reflect this change also

tonhuisman, 2023-09-02:
This seems to have been fixed now, so can be closed.

closed 2023-09-02 by TD-er