letscontrolit/ESPEasy #2787: Introduce strings as device events?

Grovkillen, 2019-11-28:
It's been a long time wish to have the possibility to import strings alongside the device values. Lets discuss this in a more organized manner.

We have the possibility to import floats using event values, MQTT import and P2P. We don't want to use the value posts for anything else than numerical values, as in something that can be plotted in a graph. But strings can be very useful in other scenarios.

So how should we go about this? Could we use the event call to pass along strings to be used in rules.

Event,Name=text

Or should it be a separate command?

This is an open discussion so please fill in.

TD-er, 2019-11-28:
One option could be to use the %eventvalue% parameter to also contain a string value, so you could for example display a string received from MQTT import.
Also if it is part of an event, you can also use the value as a trigger for rules.

Grovkillen, 2019-11-28:
Perhaps an event value parsed as a string could be used in rules under the %eventstring% so that you could use the same event for either floats or strings.

This scenario:

Event, UpdateOled="good morning"

```
On UpdateOled Do
If %eventstring%
 OLED,%eventstring%
EndIf
If %eventvalue%
 TimerSet,1,%eventvalue%
EndIf
EndOn
```

mackowiakp, 2019-11-29:
OMG! Dear developers! "I have a dream". A dream that 3 functionalities are implemented in ESP:
- SSH with certificates support (I know, I know - not enough resources)
- STRING AS %EVENTVALUE%
- the ability to compile ESPeasy with SD Card support (I can't do it all the time)
This would allow throwing out 95% of RPi and replacing them with a very cheap and maintenance-free (compared to Linux) solution.
If this happens, I can actually die peacefully ....

TD-er, 2019-11-29:
> If this happens, I can actually die peacefully ....

Hmm, we don't want to loose users, so let's not implement it all ;)

The string as %eventvalue% (or as  %eventstring% as Jimmy suggested) can be done via Import MQTT, but it cannot be done for all. So there are some limitations, but I recon it would be a good start.
I was looking through the code to process events and noticed this possibility for the import MQTT.
Maybe we can also have it added for more controllers later, but let's start here.

SD card support is on my watch list too, but I want to get the release out first, so I was looking into some stability issues first and then this idea came up.

sanyafifa, 2019-11-29:
Can this value be stored in a dummy device? For example, "UPS Online"

TD-er, 2019-11-29:
Nope, not in this initial way I now imagine how this can be used.
To change the output type of a plugin/task, a lot of breaking changes are needed, so that's something I want to avoid at first.
Maybe adding another variable type (like the `var#1` etc we now have) to store strings can be added I think.
But I am not entirely sure what else should be changed to add support for actually handling the string objects in rules.
I can imagine people want to do a lot with strings, like:
- concatenate strings
- Use it in pre-formatted strings like these: `"status: [str#1]"`
- Send it to plugins like display (probably via commands like it is done now)
- Get subsections of strings
- Search in strings
- Take out numerical values given some start position in the string (or start character)
- etc.

So I think it should be initially only to match events and copy a string value from an event and maybe also store it in a variable.
Allowing plugins to carry strings on their output does require a BIG change in the code.

macke4u, 2020-05-14:
Are the any new regarding this? It should be really nice to use MQTTimport to subscribe to a String value

TD-er, 2020-05-14:
Not a lot.

Currently there are 2 plugins now using String outputs (and the IR plugins are being tested with them as far as I know.
There has been a recent addition to the rules for simple string processing (substring, convert from HEX, etc.)
You can also insert numerical values in a pre-formatted string from rules to write to a display or generate URL etc.

But that's just about it I'm afraid.

macke4u, 2020-05-15:
Thanks for fast replay. Which plugin is using string as output? 

Are there any documentation how to use per-formatted strings to be shown on a display?

Any plans to get string support from MQTTimport?

TD-er, 2020-05-15:
[Documentation on String operations in rules](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=hex#string-formatting-and-interpreting)

Also Jimmy has used some pre-formatting in this folder with rules benchmark tests: https://github.com/letscontrolit/ESPEasy/tree/mega/test/benchmark

Adding support for strings is quite a lot of work, so I plan on doing that in steps.
Right now these use strings as output:
- [_P087_SerialProxy.ino](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P087_SerialProxy.ino)
- [_P094_CULReader.ino](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P094_CULReader.ino)

Just look for `SENSOR_TYPE_STRING` in the source code.

String import does add a whole lot of extra things to do.
For example we need to do something with it, so we must interpret the strings to get values, commands or events out of it.
And/or we may want to simply copy the string value, as those 2 plugins do.
N.B. those plugins can send data to a connected controller, but not all controllers support strings.  At least MQTT controllers can handle strings.

Also a very simple unanswered question is, how many strings should a plugin deliver? Right now it is limited to 1 and the value can also not be stored in RTC to survive a reboot.
It can not (yet) be queried in rules via the `%eventvalue%`, nor via the `[taskname#taskvar]` construct.

I think using it via the `%eventvalue%` in rules is the logical next step.
This then opens the suggestion to let _MQTT import_ just send incoming string type values only via events.

macke4u, 2020-05-15:
Thanks! I will read the documentaion to decide which way to go...

closed 2022-04-22 by TD-er