letscontrolit/ESPEasy #2790: ESP32 could not prepare wifi

henkiejan1, 2019-11-30:
I just uploaded the latest release with visual code and platformio to a ESP32. It´s the custom_ESP32_4M316k version. Uploading goes ok. But when i open serial i read this:

[code]ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
�U58 : Info  : 

INIT : Booting version:  (ESP32 SDK v3.2.3-14-gd3e562907)
58 : Info  : INIT : Free RAM:292608
59 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
69 : Info  : FS   : Mounting...
105 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
125 : Info  : CRC  : SecuritySettings CRC   ...OK 
198 : Info  : INIT : Free RAM:286972
199 : Info  : INIT : I2C
200 : Info  : INIT : SPI not enabled
216 : Info  : INFO : Plugins: 13 [Normal] (ESP32 SDK v3.2.3-14-gd3e562907)
217 : Error : WIFI : No valid wifi settings
318 : Info  : WIFI : Set WiFi to AP
1170 : Info  : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
1172 : Error : WIFI : Could not prepare WiFi!
1177 : Info  : OTA  : Arduino OTA enabled on port 3232
2519 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 219372 WiFiStatus 255
32516 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 219572 WiFiStatus 255
[/code]

TD-er, 2019-11-30:
There are no settings on the freshly installed ESP, so that's perfectly normal.
It also tells you how to setup the unit.
It is running an access point with the name "ESP_Easy_0"
So then you connect to it via your mobile. The default password is "configesp" (if I'm not mistaken)
Then you will be offered a "login prompt", similar to when you connect to a public network like in a hotel or large store.
This will redirect you to a page on `http://192.168.4.1/setup`

There you proceed with the config and save.

You can also enter the credentials via the serial port.
See the [command reference](https://espeasy.readthedocs.io/en/latest/Reference/Command.html) and look for `WifiKey` and `WifiSSID`

For further info, please see the [forum](https://www.letscontrolit.com/forum/search.php?search_id=newposts).

This is not a bug, but a user question on how to operate ESPEasy, so please use the forum for that.

tonhuisman, 2022-04-17:
This support issue doesn't have a response for quite some time, can probably be closed

closed 2022-04-17 by TD-er