letscontrolit/ESPEasy #2791: Taskrun in rules not recognized

PPatou, 2019-12-01:
Hello friends,
I am using actually ESP_Easy_mega-20191116_normal_ESP8266_4M1M.bin
I increment/decrement a setpoint by push button Up and Down in var 1
I also display this set point on Oled Frame display
The refresh rate of the display is set to 30 sec because it shows temperature data
But when setpoint is changed by Up / down it should refresh display immediately.
I wanted to use Taskrun command in rules like this

Code: 

On Up#State=1 do
 if [var#1]<=[Var#7] // compare if not above max value
 let,1,[var#1]+0.5
 taskrun,8
 endif
endon

But this command reboot the esp ?
If I give this command in the command line no reboot

Of course I can program refresh rate of display every sec but this consume cpu ressource
Is this a possible bug ?
Patou

TD-er, 2019-12-01:
Well it should not reboot.
What plugin is located at task index 8?

PPatou, 2019-12-01:
The display OLED Framed
[image]

TD-er, 2019-12-01:
Hmm I tried it here on my test board and when I do a `taskrun,1` on my setup (OLED framed is as position 1 here), it does switch nicely to the next page.

Can you show what is selected on the task configuration?
Do you have scrolling enabled, or is the update instant?

TD-er, 2019-12-01:
Can you try the test build I made for this PR: https://github.com/letscontrolit/ESPEasy/pull/2785  ?  (linked in the last comment)

PPatou, 2019-12-01:
Here is the task 8 detail
[image]

PPatou, 2019-12-01:
I tried to put the Oled display in task 1 this make no difference

TD-er, 2019-12-01:
Nope that should not make any difference. That remark was just to illustrate why I used 1 instead of 8 you used.

Can you try the test build I linked?

PPatou, 2019-12-01:
I will do it to-morrow it is quiet late now
Have a good night

PPatou, 2019-12-01:
Still awake ...
Tested with 
ESP_Easy_mega-20191127-14-PR_2785_normal_ESP8266_4M1M.bin
No more crash and reboot
Will check more to-morrow and report
Have a good night

PPatou, 2019-12-01:
Oups this bin file does not support the command OLEDFRAMEDCMD ! 
[image]

TD-er, 2019-12-01:
What is the exact syntax you try to use?

TD-er, 2019-12-01:
I do see the command must be lower case, which may be a bit inconvenient.

But `oledframedcmd` should still work.

TD-er, 2019-12-01:
> Oups this bin file does not support the command OLEDFRAMEDCMD !
> [image]

OK, found what's going on here.
The problem is that the commands for internal commands and plugin commands are dealt with differently.
The log for "Command unknown" is generated in the function handling the internal commands.
But the command is handled in the plugin.

I will make it more uniform and more clear whether a command is handled or not.

TD-er, 2019-12-02:
Fixed the issue with the logs about supposedly "unknown command" when it is a plugin command.

tonhuisman, 2023-09-02:
This seems no longer relevant, so can be closed.

closed 2023-09-02 by TD-er