letscontrolit/ESPEasy #2794: ESP32 webserver does not reply anymore after WiFi disconnect/reconnect

TD-er, 2019-12-02:
Steps to reproduce:
- Press Disconnect button in web interface

Device does react to pings, MQTT still works, but port 80 does not reply anymore.

chunter1, 2019-12-08:
I guess the same issue applies to ESP8266 modules aswell since a lot of my modules are hanging after wifi disconnect now.
Which espeasy version is the last known to be good?

TD-er, 2019-12-08:
> Which espeasy version is the last known to be good?

Other question: With which version do you experience this behavior?
A lot of users report one of the last builds from November is working great.

chunter1, 2019-12-08:
Version used is:

> Build: 20104  - Mega
> System Libraries: ESP82xx Core 2_6_0, NONOS SDK 2.2.2-dev(bb83b9b), LWIP: 2.1.2 PUYA support
> Git Build: mega-20191116
> Plugins: 46 [Normal]

Maybe it was an issue with the AP, however i never had issues with it before and even 2 resets of the AP did not change the strage behaviour that the webinterface of several units was not reachable anymore but all modules were pingable. :/

TD-er, 2019-12-08:
> strage behaviour that the webinterface of several units was not reachable anymore but all modules were pingable.

That's indeed exactly as with the ESP32 units.
I have not seen that happening on ESP8266 units.

CNC-Maniac, 2019-12-30:
Sorry, but I have exactly the same issues with the 8266 devices, using latest version (20191208).

tonhuisman, 2022-05-07:
This seems to be resolved since the last response before this one, so can probably be closed

closed 2022-05-07 by TD-er