letscontrolit/ESPEasy #2795: System Info IP  only returns 1st byte

s0170071, 2019-12-02:
when I use the Generic System Info device, I can choose the IP octets. 
No matter what I choose, it only delivers the first (192) in the devices page.

TD-er, 2019-12-02:
Hmm first quick test on my ESP32 node I was working on showed me the correct octet for IP4
Have to check on an ESP82xx unit.

clumsy-stefan, 2019-12-02:
Just FYI... works on all my nodes, incl. esp32:
[image]

thomastech, 2019-12-04:
Me too: I tried it on a recent Mega PR on a NodeMCU (ESP8266 4M) and it is working correctly. See snapshot.

[image]

I suggest you post of snapshot of the device's configuration page that shows all the System Info settings for the four octets. Include a snapshot of the device summary page too.

- Thomas

s0170071, 2019-12-07:
This is strange. Maybe config related. 
I just created a second system info device and there it works. 
No one device gives the correct octet, the other one still fails.

[image]

s0170071, 2019-12-07:
got it:  I use as formula 192.168.1.%value%,  the result is 192
What I wanted was "192.168.1.25"

TD-er, 2019-12-07:
Hmm, sounds like it will interpret the result as a float (192.168 if rounded to 3 decimals)
You want it to be handled as a string, but we don't have a method to force something as a string yet.

thomastech, 2019-12-07:
> I just created a second system info device and there it works.
No one device gives the correct octet (task6 value4), the other one still fails (task3, value3).

Sounds like plugin corruption. Delete the bad system info plugin and recreate it. With a bit of luck that might restore proper operation. 

> Hmm, sounds like it will interpret the result as a float (192.168 if rounded to 3 decimals)

Using extraneous float precision just adds more zeros to the value, such as Ip1 = 192.000.

> got it: I use as formula 192.168.1.%value%, the result is 192. What I wanted was "192.168.1.25"
Without formula the result is 25

Using the Ip4 value is just the fourth octet, not a full IPv4 type IP address. If the intent here is to acquire the full IP address then the System Info plugin would need vars for each of the four IP octets.  These can be combined in a rule to create the entire IP address.

- Thomas

TD-er, 2019-12-07:
Or just use `%ip%` if you need the complete IP.

> Using extraneous float precision just adds more zeros to the value, such as Ip1 = 192.000.

I was thinking about the string `192.168.1.1`, which is only a valid numerical upto the second `.`
So it could be parsed as a float into 192.168
Depends on where it is used. A formula returns a string which can be interpreted as a float.

thomastech, 2019-12-07:
> Or just use %ip% if you need the complete IP.

That's the ticket. :)

- Thomas

tonhuisman, 2023-09-02:
This seems to be resolved, so can be closed.

closed 2023-09-02 by TD-er