letscontrolit/ESPEasy #2801: Cannot build minimal_core_261_ESP8266_1M_OTA

2bontb, 2019-12-05:
### Summarize of the problem/feature request
I am vary happy with the core 2_6_1 improvement for the Wifi available in the normal 4M1 builds in package mega-20191130. All my devices (Wemos, Sonoff POW) are running much more stable on WiFi.

However there is no 1M  OTA build available for my Sonof basics.
So I tried via Platformio to build a minimal_core_261_1M_OTA, but this build is failing with the following error:

Processing * in minimal_core_261_ESP8266_1M_OTA environment
-------------------------------------------------------------------------------
Building...
PlatformManager: Installing platform-espressif8266
git version 2.21.0.windows.1
Cloning into 'C:\Users\JannyenRoel\.platformio\platforms\_tmp_installing-2xxpi3-package'...

warning: Could not find remote branch core_2_6_1 to clone.
fatal: Remote branch core_2_6_1 not found in upstream origin
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'core_2_6_1', 'https://github.com/Jason2866/platform-espressif8266.git', 
'C:\\Users\\JannyenRoel\\.platformio\\platforms\\_tmp_installing-2xxpi3-package']
Unable to execute: platformio
Make sure cmd:'platformio' and cwd:'D:\ESPEasy_mega-20191130\source' exists and have correct access permissions.
Binaries are found in these folders: C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\
iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\Intel(R) Management Engine Componen
ts\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel
\Intel(R) Management Engine Components\IPT;C:\Program Files\Calibre2\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\LLVM\bin;C:\
Program Files\Git\cmd;C:\Users\JannyenRoel\AppData\Local\Microsoft\WindowsApps;C:\Users\JannyenRoel\AppData\Local\atom\bin;%USERPROFILE%\AppData\Local\Microsoft\WindowsApp
s;

Can you help me to get a suc6ful build, or provide this version in the package?

### Expected behavior
Build should be suc6ful

### Actual behavior
Error message.

### Steps to reproduce
See above

TD-er, 2019-12-05:
Have you been successful before on this computer to build ESPEasy with PlatformIO?

The reason I ask is that it seems like you have not installed the Git command line tools.
See [PlatformIO pre-requisites](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#platformio-prerequisites)

2bontb, 2019-12-06:
Yes I have build at the same time the 2_4_2 version as a check.
also if I remove all packages, and do a clean you see it installs via GIT:
Processing custom_ESP8266_4M1M (platform: https://github.com/Jason2866/platform-espressif8266.git#feature/stage; board: esp12e; framework: arduino)
-------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading...
Unpacking...
toolchain-xtensa @ 2.40802.191122 has been successfully installed!
PackageManager: Installing tool-esptool @ <2

Downloading...
Unpacking...
tool-esptool @ 1.413.0 has been successfully installed!
PackageManager: Installing tool-esptoolpy @ ~1.20800.0
Downloading...
Unpacking...
tool-esptoolpy @ 1.20800.0 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266
git version 2.21.0.windows.1
Cloning into 'C:\Users\JannyenRoel\.platformio\packages\_tmp_installing-bxfjxs-package'...
Checking out files: 100% (1359/1359), done.
Submodule 'libraries/ESP8266SdFat' (https://github.com/earlephilhower/ESP8266SdFat.git) registered for path 'libraries/ESP8266SdFat'

Also if I use branch: feature/stage for the build, it will download the package. If I use the branch core_2_6_1, it cannot find it. Looking at github, also this branch is not shown for me at: https://github.com/Jason2866/platform-espressif8266

So I think the setup in the platformio_core_defs.ini may be wrong?

TD-er, 2019-12-06:
Where is the Jason2866 still linked in the source tree?

If you fetch the latest sources from Git, then there is no reference anymore to this repo.

But you will run into another issue, which is really, really frustrating right now as it also prevents me to do anything.

See: https://github.com/esp8266/Arduino/issues/6832

2bontb, 2019-12-06:
I used the source that was included in the  latest package of the daily build. I assumed that this was also the source that was used tobuild also the SONOFF POW with core 2.6.1
I see that the current GIT changed to 2.6.2, (2.6.1. is commented out) I will try that one. But last days we did not get a daily buil. So I presume that I will walk into the problems you indicated above.
I will go back to the 20190827 build. Becuase the Sonoff basics that I upgraded with the changed 2.6.1 path did crash the basics.

I think this discussion can be closed. Unfortunate, because the 2.6.1 of the POW are working very well.

2bontb, 2019-12-07:
Note: I cloned the github of 6-12-2019. Could compile minimal core 261 8285 and use it on the basic's.
Do not understand why the 8266 version did not work, but I noticed it before that may Basics are more stable on the 8285.
Also sdk3 did not work for me.
Now see how stable it is coming days.

closed 2019-12-07 by 2bontb

TD-er, 2019-12-07:
> Also sdk3 did not work for me.

Not only for you, it is problematic for anyone right now.