letscontrolit/ESPEasy #2805: It is not possible to parse a string on http:x.x.x.x/control?cmd=event,show=test

2bontb, 2019-12-08:
In the past I used events to fill the display (device = Display - LCD2004)
http://10.2.1.119/control?cmd=event,show=Display Value
For the display I had the following rule:
// Display the data
 on show do
   lcdcmd,on
   lcdcmd,clear
   lcd,1,1,%eventvalue%
   gpio,13,1
   timerSet,1,2
   timerSet,2,10
 endon

 //beeper
 On Rules#Timer=1 do 
   gpio,13,0
 endon

 // lcd
 On Rules#Timer=2 do 
   lcdcmd,off
 endon

Using release 20191130 I observe two things
1. Only numbers can be passed on the event,show=xxxxx; When not a number is used, the event does not trigger anymore.
2. for the lcd command lcd,x,y,text the text need to be quoted, else only the first word is shown. I fixed this by adding double quotes in the rules around the %eventvalue%.

Is there a other way to trigger an event, and pass a string to the event to execute the commands as displayed in above rule? 
I could not find it in the rules examples: https://espeasy.readthedocs.io/en/latest/Rules/Rules.html

TD-er, 2019-12-08:
Please see also this sticky issue I made with regards to the recent change in rules parsing:
https://github.com/letscontrolit/ESPEasy/issues/2724

TD-er, 2019-12-08:
I just entered the command you mentioned in the title via the web interface on Tools => command field.

This resulted in this url: [http://192.168.1.75/tools?cmd=event%2Cshow%3Dtest](http://192.168.1.75/tools?cmd=event%2Cshow%3Dtest)

Can you try that syntax for the command?

2bontb, 2019-12-08:
I did: http://10.2.1.119/tools?cmd=event%2Cshow%3Dtest

Resulted in no trigger
8733981: WD : Uptime 146 ConnectFailures 0 FreeMem 18968 WiFiStatus 3
8748426: EVENT: Clock#Time=Sun,19:40
8763981: WD : Uptime 146 ConnectFailures 0 FreeMem 18952 WiFiStatus 3
8771797: Command: event
8771797: EVENT: show=test
<nothing>

However:  http://10.2.1.119/tools?cmd=event%2Cshow%3D123

Resulted in trigger
8836987: Command: event
8836988: EVENT: show=123
8836994: ACT : lcdcmd,on
8836997: ACT : lcdcmd,clear
8837003: ACT : gpio,13,1
8837006: SW : GPIO 13 Set to 1
8837007: ACT : timerSet,1,2
8837015: Command: timerset
8837016: ACT : timerSet,2,10
8837024: Command: timerset

2bontb, 2019-12-08:
Also adding quotes did not trigger the event.
9028198: Command: event
9028198: EVENT: show='test'

TD-er, 2019-12-08:
Please note that when you send the commands via http, you have to escape some characters.
Try your command via the web interface (Tools => command field) and see in the URL bar of the browser into what it was changed.
You have to use that specific one via HTTP URLs.

TD-er, 2019-12-08:
Right now we don't support strings as event values.
There are commands to set text on displays, but not through events. (yet)

2bontb, 2019-12-09:
I reprogrammed my controler to trigger the event, and after that I send the diplay string in separate lcd command.
I leave it up to you to keep this open as feature request, else you can close the ticket.
Thanks for the support.

TD-er, 2019-12-09:
If it remains as a feature request, then it is a duplicate of this one: https://github.com/letscontrolit/ESPEasy/issues/2787
So then it can be closed.

closed 2019-12-09 by TD-er