letscontrolit/ESPEasy #2807: oledframedcmd command not recognize

PPatou, 2019-12-08:
Hello,
Using : ESP_Easy_mega-20191208_normal_ESP8266_4M1M.bin
The command "oledframedcmd" is not anymore recognize
This concern all the oledframedcmd :
OLEDFRAMEDCMD,<value> | on, off,   clear | 
OLEDFRAMEDCMD,<value> | high, med,   low |

TD-er, 2019-12-08:
Hmm, apparently the command has changed and I missed it when merging it. (sorry about that)
Also the documentation is lacking.

https://github.com/letscontrolit/ESPEasy/blob/5c2bf292f4ca3163082cd220f906bd4630bcb05c/src/_P036_FrameOLED.ino#L696-L727

It looks like you have to add `display` inbetween.
For example `oledframedcmd display on`

PPatou, 2019-12-08:
Ok thanks it works with oledframed,display,on ... off .. etc ...
Is it also valid for the Oled normal display ?

TD-er, 2019-12-08:
As far as I know that plugin was not changed recently.  (I did not change the Framed OLED one myself)

PPatou, 2019-12-08:
Hello
I have adapted my document to this new command. Please correct it if there are still errors before publishing it. 
Please fell free to use it in your documentation
https://drive.google.com/open?id=1EHUaNBjns5ZGmu8DG7qS2XKVkPGSCj7O
Good night
Patou

TD-er, 2019-12-08:
Well it is incomplete, as you now can also set texts on a line for the framed oled plugin.

For example `oledframedcmd,1,"text"` where `1` is the line number.

closed 2019-12-09 by PPatou