letscontrolit/ESPEasy #2816: Issue with reading from 2 MAX31855 Thermocouples

mawa85, 2019-12-14:
I have a Wemos D1 Mini with ESPEasy Release mega-20191208 and 2 MAX31855 Thermocouples connected (GPIO13 & GPIO2) I want to read both temperature values (Wifi BBQ Thermometer).

There is a strange issue: when I setup both devices, in web interface , both values are measured correctly. With the next reading, the first probe receives 0.0 and keeps that measure, The second probe continues to measure correctly. When I go into the devices menu for probe 1 and click submit again, the correct value is shown again until the next automatic reading is trigered (0.0 then again)

I already changed the GPIO but without luck

TD-er, 2019-12-14:
What plugin are you using?

When looking at the code of Plugin 039: Thermocouple (MAX6675 / MAX31855), then it does seem to be a SPI sensor:
```c++
// Wiring
// https://de.wikipedia.org/wiki/Serial_Peripheral_Interface
// You need an ESP8266 device with accessible SPI Pins. These are:
// Name   Description     GPIO      NodeMCU   Notes
// MOSI   Master Output   GPIO13    D7        Not used (No Data sending to MAX)
// MISO   Master Input    GPIO12    D6        Hardware SPI
// SCK    Clock Output    GPIO14    D5        Hardware SPI
// CS     Chip Select     GPIO15    D8        Hardware SPI (CS is configurable through the web interface)
```
This means all should be set the same, only the chip select should differ per task.

But still, not all plugins are written to be used with multiple instances and maybe this one is one of those.

uzi18, 2019-12-14:
@TD-er this one should work with multiple instances

uzi18, 2019-12-14:
@mawa85 please provide DEBUG log

mawa85, 2019-12-14:
Exactly this one, you only can choose CS PIN in Frontend for the Probes, SCK & MISO is used by all sensors.

Strange thing is, that it only faults on automatic update and works when I manually update in the devices section

uzi18, 2019-12-14:
@mawa85 and put here screenshots of tasks configuration

mawa85, 2019-12-14:
The log on debug-dev.
First update was manual triggered, second update automatically after 60 seconds, receives 0 value. As soon as I trigger manually again, I get the correct values

`3858650: Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 2 4: 'TDE' length: 2 5: 'taskde
3858653: SaveToFile: free stack: 3136
3858684: FILE : Saved config.dat
3858685: SaveToFile: free stack after: 3136
3858695: SaveToFile: free stack: 3392
3858725: FILE : Saved config.dat
3858726: SaveToFile: free stack after: 3392
3858738: SaveToFile: free stack: 3328
3858799: FILE : Saved config.dat
3858800: SaveToFile: free stack after: 3328
3858810: P039 : SPI Init
3858907: P039 : MAX31855 : RAW - BIN:111000010000000011011100000 HEX:70806e0 DEC:117966560
3858908: P039 : Temperature 112.50
3866845: P039 : MAX31855 : RAW - BIN:100100011000000011000100000 HEX:48c0620 DEC:76285472
3866846: P039 : Temperature 72.75
3872594: WD : Uptime 65 ConnectFailures 0 FreeMem 19256 WiFiStatus 3
3902594: WD : Uptime 65 ConnectFailures 0 FreeMem 19256 WiFiStatus 3
3918917: P039 : MAX31855 : RAW - BIN:0 HEX:0 DEC:0
3918918: P039 : Temperature 0.00
3926846: P039 : MAX31855 : RAW - BIN:100100011000000011001100000 HEX:48c0660 DEC:76285536
3926847: P039 : Temperature 72.75
3932594: WD : Uptime 66 ConnectFailures 0 FreeMem 19208 WiFiStatus 3`

mawa85, 2019-12-14:
3858650: Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 2 4: 'TDE' length: 2 5: 'taskde
3858653: SaveToFile: free stack: 3136
3858684: FILE : Saved config.dat
3858685: SaveToFile: free stack after: 3136
3858695: SaveToFile: free stack: 3392
3858725: FILE : Saved config.dat
3858726: SaveToFile: free stack after: 3392
3858738: SaveToFile: free stack: 3328
3858799: FILE : Saved config.dat
3858800: SaveToFile: free stack after: 3328
3858810: P039 : SPI Init
3858907: P039 : MAX31855 : RAW - BIN:111000010000000011011100000 HEX:70806e0 DEC:117966560
3858908: P039 : Temperature 112.50
3866845: P039 : MAX31855 : RAW - BIN:100100011000000011000100000 HEX:48c0620 DEC:76285472
3866846: P039 : Temperature 72.75
3872594: WD : Uptime 65 ConnectFailures 0 FreeMem 19256 WiFiStatus 3
3902594: WD : Uptime 65 ConnectFailures 0 FreeMem 19256 WiFiStatus 3
3918917: P039 : MAX31855 : RAW - BIN:0 HEX:0 DEC:0
3918918: P039 : Temperature 0.00
3926846: P039 : MAX31855 : RAW - BIN:100100011000000011001100000 HEX:48c0660 DEC:76285536
3926847: P039 : Temperature 72.75
3932594: WD : Uptime 66 ConnectFailures 0 FreeMem 19208 WiFiStatus 3
3986845: P039 : MAX31855 : RAW - BIN:100100011000000011001100000 HEX:48c0660 DEC:76285536
3986846: P039 : Temperature 72.75
3992594: WD : Uptime 67 ConnectFailures 0 FreeMem 19192 WiFiStatus 3
4022594: WD : Uptime 67 ConnectFailures 0 FreeMem 17624 WiFiStatus 3
4038916: P039 : MAX31855 : RAW - BIN:0 HEX:0 DEC:0
4038917: P039 : Temperature 0.00
4046845: P039 : MAX31855 : RAW - BIN:100100010000000011000100000 HEX:4880620 DEC:76023328
4046846: P039 : Temperature 72.50
4052594: WD : Uptime 68 ConnectFailures 0 FreeMem 19160 WiFiStatus 3

mawa85, 2019-12-14:
Only differnce from the logs is the **SPI Init** when I trigger manually, could this solve the issue?

uzi18, 2019-12-14:
what name of file you used to flash esp module?

mawa85, 2019-12-14:
ESP_Easy_mega-20191208_normal_ESP8266_4M1M.bin

uzi18, 2019-12-15:
[firmware.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3964492/firmware.bin.zip)
show me debug log from this build

mawa85, 2019-12-15:
1018886: SaveToFile: free stack: 3328
1018920: FILE : Saved security.dat
1018921: SaveToFile: free stack after: 3328
1018959: P039 : SPI Init
1019153: P039 : MAX31855 : RAW - BIN:1011111000001011111010000 HEX:17c17d0 DEC:24909776
1019153: P039 : MAX31855 : PIN :13
1019154: P039 : Temperature 23.75
1024358: Webserver args: 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 2 4: 'TDE' length: 2 5: 'taskde
1024361: SaveToFile: free stack: 3088
1024392: FILE : Saved config.dat
1024393: SaveToFile: free stack after: 3088
1024403: SaveToFile: free stack: 3344
1024433: FILE : Saved config.dat
1024434: SaveToFile: free stack after: 3344
1024446: SaveToFile: free stack: 3280
1024505: FILE : Saved config.dat
1024505: SaveToFile: free stack after: 3280
1024516: P039 : SPI Init
1024629: P039 : MAX31855 : RAW - BIN:1011111000001011111000000 HEX:17c17c0 DEC:24909760
1024629: P039 : MAX31855 : PIN :13
1024630: P039 : Temperature 23.75
1038640: WD : Uptime 17 ConnectFailures 0 FreeMem 18304 WiFiStatus 3
1068640: WD : Uptime 18 ConnectFailures 0 FreeMem 19840 WiFiStatus 3
1069628: P039 : MAX31855 : RAW - BIN:1010110000001011100110000 HEX:1581730 DEC:22550320
1069629: P039 : MAX31855 : PIN :2
1069629: P039 : Temperature 21.50
1084641: P039 : MAX31855 : RAW - BIN:0 HEX:0 DEC:0
1084641: P039 : MAX31855 : PIN :13
1084642: P039 : Temperature 0.00
1098641: WD : Uptime 18 ConnectFailures 0 FreeMem 19808 WiFiStatus 3
1128641: WD : Uptime 19 ConnectFailures 0 FreeMem 19792 WiFiStatus 3
1129628: P039 : MAX31855 : RAW - BIN:1011110000001011101000000 HEX:1781740 DEC:24647488
1129628: P039 : MAX31855 : PIN :2
1129629: P039 : Temperature 23.50