letscontrolit/ESPEasy #2817: Feature request: Forced ABC calibration for Senseair S8 or Sunrise

fluppie, 2019-12-15:
Is it possible to implement this? I saw changing the ABC period was possible in the past but due too a bug, it was disabled.
For the Sunrise S11 I saw you could force ABC correction when unit is in fresh air / 400 ppm. See page 17 https://rmtplusstoragesenseair.blob.core.windows.net/docs/Market/publicerat/TDE7318.pdf
Any idea if this is hard/a lot of work to implement?

TD-er, 2019-12-15:
It is not hard work, but it is very likely you will not have the correct environment when running the ABC and thus ruin the existing calibration.
For the S8 there is still a bug in the firmware, so for that we need to have a good look at how to prevent the unit to lock up.
Maybe we can also check for a firmware revision and only enable it for the ones that do not have this bug.

fluppie, 2019-12-15:
I was thinking to take the sensor outside for 10-15 min and then reset to 400ppm. Or is this not a good approach?

Red = Senseair S8
Green = Senseair S11
Sensors are ofcourse in the same room and within 5-10cm of eachother.
[image]

The S8 is reporting values of 300-320 with the windows open and the S11 500-520. So I think the ABC for both needs to be initiated. Ofcourse we can wait for a long time, but it would be nicer to force ABC adjusting when outside or with the windows wide open for 30-60min.

TD-er, 2019-12-15:
The measured CO2 value is depending on the temp/humidity and also the outside CO2 value is not guaranteed to be 400 ppm.
For example when you are near traffic, the values may be (a lot) higher, or when there is no sunlight but plants or trees in the neighborhood, then the actual CO2 value can be higher.

Direct sunlight on the sensor will also affect the readings.

So like I said it is very likely you will not have the correct environmental conditions to truly reset them to 400 ppm.
This offset you are seeing is quite odd.
I will ask the people I know at SenseAir about the best procedure to force the sensor at the lowest recorded value in some interval.

Grovkillen, 2019-12-15:
I would trust the ABC and not rush things. It's been proven to work very reliable for the S8 which has been on the market for many years. But if you want to tinker I'd suggest we have the plug-in also do a reset of the ABC counter at the same time since the S8 got that overflow bug.

fluppie, 2020-12-26:
Was playing around with the sensors to make a nicer interface into HA. Still seeing an offset between S8 and S11, still being in the same room, 5-10cm apart.
[image]

The "old" S8's in the living room are almost exactly the same value :), also 10cm apart.

Grovkillen, 2020-12-26:
This is the main reason I don't use ppm because it makes 678 particles per
1 000 000 and 745 particles per 1 000 000 seems like huge difference.

I convert the ppm to what I call Rel%CO2 which is a range of 350 ppm (0%)
to 2 000 ppm (100%). This means that your values could also be more then
100% in really bad environments and negative values in super fresh air.

Grovkillen, 2020-12-26:
Think of the smell of a fart... In a confined space it'll smell more for
some and less for others. This is because the smelly particles need some
time to get mixed with the rest of the air.

Den lör 26 dec. 2020 14:18Jimmy Westberg <jimmy@grovkillen.com> skrev:

fluppie, 2020-12-26:
I agree, but I've opened the window (270x120cm) completely and still big difference in the measurements. The S11 reports below 400?
[image]

fluppie, 2020-12-26:
The firmware of the S8's is 30 11 2019, because it's very stable, gets easily 30+ days uptime. The S11 is running 30 11 2020. So a quite recent release. But I don't think this should make a difference?

Grovkillen, 2020-12-26:
Is it perhaps more influenced by the temperature? The S11 use an LED and
the S8 is using a lamp.

fluppie, 2020-12-26:
Maybe, 11.6 for the S11 vs 14.3 for the S8. Room temperature sensor in the KNX Switch says 14.6 deg.

Grovkillen, 2020-12-26:
Let it be and the ABC will make them more and more close to each other. A month from now and they will be the same.

fluppie, 2020-12-26:
Does a power cycle reset the ABC calibration? They're in this room, in this exact position since last year :-)...

Grovkillen, 2020-12-26:
No the ABC is based on the number of hours "on". So it will not care about
power cycles. What is the exact reason on the need to have them show
exactly the same?

fluppie, 2020-12-26:
Long term test S8 vs S11

Grovkillen, 2020-12-26:
Ok I guess you need to have more units to do a significant amount of samples to draw any conclusions. The values you have in the graph tells us that they are pretty stable and only have a linear offset to each other. If you use them both and average their values you got a super great sensor.

TD-er, 2020-12-26:
Do the sensors get some external influences like direct sunlight?
Or something radiation IR light?

Also could it be that one of the sensors (or both) may have had some dirt on them which may have affected the ABC routine on one of them?