letscontrolit/ESPEasy #2818: [Generic HTTP Advanced controller] does not accept values from other tasks e.g. [dht22#temp]

meekm, 2019-12-15:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

meekm, 2019-12-15:
[Generic HTTP Advanced controller] does not accept values form other tasks e.g. [dht22#temp]

meekm, 2019-12-15:
I want to send 8 sensor values to api.thingspeak.com using one POST request, but i can't access other tasks.

TD-er, 2019-12-15:
Indeed the http header and the body are not fed to the parse function to replace internals.
That should be an easy fix.

TD-er, 2019-12-15:
Looking at the code, and it seems like this controller does use a quite specific format.
Something between `%1%` and `%/1%` does refer to the first parameter of that task calling the controller.
But there is also a function to delete not needed values.

Can you check what happens when you place your http body or header string between `%1%` and `%/1%`  ?

TD-er, 2019-12-15:
Can you try the test build I linked here: https://github.com/letscontrolit/ESPEasy/pull/2820 ?

PR #2820 (2019-12-18): [Parse Templates] Allow to refer to variables in controllers (#2818)
Fixes: #2818
commit c0eb4ec2 (2019-12-18): [Parse Templates] Allow to refer to variables in controllers (#2818)

Fixes: #2818
closed 2019-12-18 by TD-er