letscontrolit/ESPEasy #2819: Minimal ESP8266 build 20191208 rules get corrupted when saved

jayrisko, 2019-12-15:
When rules of several lines are entered and saved using a "minimal" build (i.e. 1M) on ESP8266 (Sonoff device) they become corrupted at the top of the rules page.  

Tried "blanking" and reflashing - 2 devices are acting the same.

TD-er, 2019-12-15:
Do you get a red dialog when saving?
It can be that is only part of the last build, but I also noticed the issue and added some checks for it, so at least the existing rules will not be deleted. (but the last changes will be lost though)

The bug seems to be related to the WebServer code, handling the decoding of the POST arguments.
See my issue in esp8266/Arduino: https://github.com/esp8266/Arduino/issues/6884

jayrisko, 2019-12-16:
> See my issue in esp8266/Arduino: esp8266/Arduino#6884

No red dialog when saving - no indication anything is amiss until
re-entering the rules area, then some random characters are at the top,
with some portion of up to several lines of the rules deleted.

Budman1758, 2019-12-16:
Possibly related?
[https://github.com/letscontrolit/ESPEasy/issues/1296](url)

TD-er, 2019-12-16:
> Possibly related?
> #1296

It could be.

jayrisko, 2019-12-16:
Read details on #1296 and this seems similar, except that a save messes up
only a few (5-10?) characters at the top, and inserts several unprintable
or non-ascii characters. A repeated save on an already messed up rule set
results in corruption further downwards by up to an additional line or
two...

PR #2921 (2020-03-23): Rules save fix
Now the rules are uploaded as a (virtual) file instead of form data. The previous way resulted in corrupted rules files if the unit didn't get the full POST body.

Should fix #2896
Should fix: #2819
Should fix: #2729
Should fix: #1296
closed 2020-03-23 by TD-er