letscontrolit/ESPEasy #2821: reboot on static ip on

stefbo27, 2019-12-16:
After assigning a static IP and manually rebooting the latest git version (11.12.  345cf2b7ec16af978cb318427fb72fa10f328311) goes into reboot loop.
build target: dev_ESP8266_4M1M on esp8266-12e 
The device was stable with previous versions so I do not assume a hardware issue.
Here is the serial log 

```
INIT : Booting version: mega 2019-12-12 (ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
90 : Info  : INIT : Free RAM:30640
91 : Info  : INIT : Warm boot #8 Last Task: Background Task - Restart Reason: Exception
93 : Info  : FS   : Mounting...
117 : Info  : FS   : Mount successful, used 75802 bytes of 957314
135 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
145 : Info  : CRC  : SecuritySettings CRC   ...OK
251 : Info  : INIT : Free RAM:27464
252 : Info  : INIT : I2C
253 : Info  : INIT : SPI not enabled
1619 : Info  : INFO : Plugins: 82 [Normal] [Testing] [Development] (ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
1723 : Info  : WIFI : Set WiFi to STA
1756 : Info  : WIFI  : Start network scan
1758 : Info  : WIFI : Connecting XYZ attempt #0

Exception (9):
epc1=0x40293ef0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff196e depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff2d60 end: 3fff2fe0 offset: 01a0
3fff2f00:  00000004 3fff47e4 3fff0dec 3fff161c
3fff2f10:  3fff15fc 3fff15dc 00000020 4010187b
3fff2f20:  3fff15fc 3fff47e4 00000002 40237449
3fff2f30:  40294910 00000000 40294910 00000000
3fff2f40:  40294910 00000000 40294910 00000000
3fff2f50:  3fff15fc 3fff15dc 3fff0b90 40250d30
3fff2f60:  3fff4700 0022002f 80ff1964 4028750c
3fff2f70:  2e310029 3ffe872c 00000002 3fff1ea4

3fff2f80:  3fff0de9 3ffe872c 3fff1964 402718f8
3fff2f90:  3fff6000 004a004f 8029005d 40293154
3fff2fa0:  3fff60ac 0080008f 00efeffe feefeffe
3fff2fb0:  feefeffe feefeffe feefeffe 3fff3028
3fff2fc0:  3fffdad0 00000000 3fff2fe8 40288728
3fff2fd0:  feefeffe feefeffe 3ffe8764 401007d9
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbc204a9b
~ld
�U90 : Info  :

INIT : Booting version: mega 2019-12-12 (ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
91 : Info  : INIT : Free RAM:30640
```

TD-er, 2019-12-16:
Do you have any controllers enabled?

stefbo27, 2019-12-16:
No controllers, no devices

stefbo27, 2019-12-16:
Just tried same version (I use custom settings for default values and enable Plugin P042 (Candle) based) on 7fc1d061938ccf0a4e4946516bce9e5bc5dc1841 and it works.

TD-er, 2019-12-16:
```
1756 : Info  : WIFI  : Start network scan
1758 : Info  : WIFI : Connecting XYZ attempt #0
```
Seems strange at first sight, that the scan only takes 2 msec, so I guess there may be a problem where it is not waiting for the the scan result.

TD-er, 2019-12-16:
> Just tried same version (I use custom settings for default values and enable Plugin P042 (Candle) based) on [7fc1d06](https://github.com/letscontrolit/ESPEasy/commit/7fc1d061938ccf0a4e4946516bce9e5bc5dc1841) and it works.

Can you post the Custom.h file you used?
Please make sure to remove any credentials (replace them by something like `"secret"`

stefbo27, 2019-12-16:
here is the log from working version
```
INIT : Booting version: mega 2019-12-12 (ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
83 : Info  : INIT : Free RAM:30640

84 : Info  : INIT : Cold Boot - Restart Reason: External System
86 : Info  : FS   : Mounting...
111 : Info  : FS   : Mount successful, used 75802 bytes of 957314
128 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
138 : Info  : CRC  : SecuritySettings CRC   ...OK
244 : Info  : INIT : Free RAM:27480
245 : Info  : INIT : I2C
246 : Info  : INIT : SPI not enabled
532 : Info  : INFO : Plugins: 82 [Normal] [Testing] [Development] (ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
637 : Info  : WIFI : Set WiFi to STA
670 : Info  : WIFI  : Start network scan
671 : Info  : WIFI : Connecting XYZ attempt #0
672 : Info  : IP   : Static IP : 192.168.178.110 GW: 192.168.178.1 SN: 255.255.255.0 DNS: 192.168.178.1
2324 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 22008 WiFiStatus 6
3964 : Info  : WIFI : Connected! AP: XYZ (xx:xx:xx:xx:xx:xx) Ch: 1 Duration: 3111 ms
3966 : Info  : WIFI : Static IP: 192.168.178.110 (ESP-EasySB-0) GW: 192.168.178.1 SN: 255.255.255.0   duration: 181 ms
3970 : Info  : Webserver: start
4034 : Info  : NTP  : NTP replied: delay 61 mSec Accuracy increased by 0.946 seconds
4035 : Info  : Time adjusted by 1576539019979.89 msec. Wander: 437927505.55 msec/second
4037 : Info  : Current Time Zone:  DST time start: 2019-03-31 02:00:00 offset: 120 minSTD time start: 2019-10-27 03:00:00 offset: 60 min
4228 : Info  : firstLoopConnectionsEstablished
32022 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 17568 WiFiStatus 3
```

stefbo27, 2019-12-17:
I can confirm the same behaviour without customization. I first flashed with blank_4MB image to clean all settings. Than I flashed the original 345cf2b7ec16af978cb318427fb72fa10f328311 commit. 
After configuring the WIFI I rebooted (DHCP) --> worked
Then I changed to fixed IP and reboot: --> reboot-loop
Flashing then (keeping config for fixed IP) image based on mega-20191208 7fc1d061938ccf0a4e4946516bce9e5bc5dc1841  --> works

TD-er, 2019-12-17:
OK, so the changes since the 8th do something with the fixed IP.

Thanks for narrowing it down. That's really helpful.

FanOfHue, 2019-12-21:
Same trouble here. Previous version from 20191027 was fine. Tried with github from 20191220. No change in settings, config or whatsoever. Results in a fast crashloop just when trying to connect to Wifi (using static IP). Had to open the box to get to the serial connector to revive the unit.

_Would also be much more convenient to have stable builds instead of nightly dev editions only._

thomastech, 2019-12-26:
I also confirm that static IP breaks recent releases. DHCP is Ok. 

See the boot loop discussion here:
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=7261&p=41649#p41647

- Thomas

bojanul, 2020-01-26:
Exception (9) boot loop with static ip is present if I build the firmware by myself in platformio. If I use the precompiled builds everything works fine. I had to go back to ESPEasy-mega-20191028 to get the working setup. I did not try all November builds.

TD-er, 2020-01-26:
@bojanul 
I can indeed reproduce this static IP stuff here on my node too, so it should be possible for me to find the bug here.

I am now (finally) back in business with respect to programming, so I will pick up this issue as soon as possible.
Today I was working on some issues where controller settings could get mangled and causing strange issues.

TD-er, 2020-02-03:
Can you try this [test build](https://www.dropbox.com/s/cxn8i7hcqtlyzpd/ESPEasy_mega-20191208-40-PR_2864.zip?dl=0) ? 
And then especially a build with "beta" in the name?

bojanul, 2020-02-04:
> Can you try this [test build](https://www.dropbox.com/s/cxn8i7hcqtlyzpd/ESPEasy_mega-20191208-40-PR_2864.zip?dl=0) ?
> And then especially a build with "beta" in the name?

Both build works fine with static IP.

TD-er, 2020-02-04:
The "beta" one should be a bit more stable (not crashing) on highly unstable network connection or when the MQTT broker is unreachable.

Thanks for testing.

It also has quite some changes with respect to controllers, so if you don't experience strange issues, I will merge it later today and make a new nightly build

bojanul, 2020-02-04:
I uploaded the test_beta_ESP8266_4M1M build to my busiest node and it looks like everything works fine.

TD-er, 2020-02-04:
If you feel it misses a beat, then please check sending the ARP packets in the advanced (experimental) settings.

stefbo27, 2020-02-04:
I tested ESP_Easy_mega-20191208-40-PR_2864_test_beta_ESP8266_4M1M.bin and ESP_Easy_mega-20191208-40-PR_2864_dev_ESP8266_4M1M.bin 
Both work with static IP.
Just for your info:
I compared with a version based on latest github commit (3b1bb1). That version does NOT work with static IP. I applied changes from 8b11057 ( call to setStaticIPconfig() at end of preoareWiFi() ) but htat does not solve the problem. maybe other changes have been made as well....

TD-er, 2020-02-04:
There have been other changes too indeed, all in this PR: https://github.com/letscontrolit/ESPEasy/pull/2864

I will merge it to mega, as it seems to fix more that it breaks ;)

closed 2020-02-04 by TD-er

stefbo27, 2020-02-04:
just downloaded latest mega. Works fine with static IP. Thanks a lot!!!

PR #2864 (2020-02-04): [Controllers] Restructure protocolIndex / controllerIndex and cplugin_ID
This was rather messy code and no checks.
It is now somewhat similar to the code for plugins defining the relation between:
- Plugin ID
- Task Index
- Device Index.

Also added the option to call CPLUGIN_EXIT and a call for 10/sec loops which we apparently didn't have until now for controllers.

Fixes: #2821
Fixes: #2866