letscontrolit/ESPEasy #2823: switch state doesn't reflect in basicUI when it chenges in ESP Easy web interface

ka-bro, 2019-12-18:
Good day!

esp easy mega-20191130
[image]
for the 2-nd switch settings are the same..

[image]

openhab 2.5.0.M3
.items
Switch	 relay1 "реле1" {channel="mqtt:topic:b0d2f1d2:relay1", autoupdate = "true"}
Switch	 relay2 "реле2" {channel="mqtt:topic:b0d2f1d2:relay2", autoupdate = "true"}

.sitemap
Switch item=relay1 icon="wallswitch"
Switch item=relay2 icon="light"

.things
Type switch : relay1 "On/Off switch" [ stateTopic="wroom-32/relay1/state", commandTopic="wroom-32/relay1/gpio/26" ]
Type switch : relay2 "On/Off switch" [ stateTopic="wroom-32/relay2/state", commandTopic="wroom-32/relay2/gpio/27" ]

All works good from basicUI. But if I change switch state from ESP Easy web interface
[image]
[image]
it doesn't reflect in basicUI
[image]
So when I reboot ESP relay stetes becomes OFF by default but in basicUI it is still ON even I refresh page in browser.
How can I get real relay status to openHAB from ESP Easy when ESP reboots?

Grovkillen, 2019-12-19:
You probably need to use a rule and not directly set GPIO since that will not trigger an event (which is what is needed for the switch plugin to trigger).

So it's not a bug but perhaps it could be done more intuitive...

Grovkillen, 2019-12-19:
Note to self: GPIO command should never be used "on its own".. only in rules and in conjunction of a gpio monitor or task run command...

TD-er, 2019-12-19:
Maybe also related to this PR: https://github.com/letscontrolit/ESPEasy/pull/2778  ???

Grovkillen, 2019-12-19:
Totally related @TD-er, I will close this one.

giig1967g, 2019-12-19:
@TD-er : I don't think so.
The issue is related to the way openHab displays the state of a GPIO.

I will post my openHab items files

Grovkillen, 2019-12-19:
The idea is that we want to take care of the GPIO commands so even if this issue are specific for basicUI the underlying problem is that GPIO isn't good to use since it'll not trigger an event.

TD-er, 2019-12-19:
As @giig1967g is the expert on GPIO stuff right now, I re-open this issue :)

giig1967g, 2019-12-19:
@ka-bro:
In order to get an update from the gpio, you first need to send this command:
```
on System#Boot
 monitor,gpio,27
endon

```

In order to update the openHab sitemap, you have to declare a Switch and a Contact item linked to the GPIO item. I use MQTT binding version 1 (not updated yet to version 2).

Example of one of my relays (change the values according to your config):
```
file = esp.items

Switch sESPT_R1 "sR1"  {http=">[ON:GET:http://192.168.1.203/control?cmd=gpio,27,1] >[OFF:GET:http://192.168.1.203/control?cmd=gpio,1,0]"}
Contact cESPT1_R1 "cR1" {mqtt="<[mqtt_broker:ESPT1/Relay/r1:state:default]"}

```

ka-bro, 2019-12-20:
Thanks to all for explanations!
I'll try this soon and report here..

tonhuisman, 2023-09-02:
This seems to be solved, and can be closed.

closed 2023-09-02 by TD-er