letscontrolit/ESPEasy #2832: Rule triggered based on SSID in range (feature request)

blb4github, 2019-12-22:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [X] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Feature request: I like to have the possibility to publish an MQTT message once a WIFI SSID is in reach and out of reach.
Via webinterface->tools->scan it's possible to get an overview of all SSID's in range. In the log I can see the number of SSID's found. I would like to trigger a rules based on the name of an SSID in range.

TD-er, 2019-12-22:
The ESP will hardly ever perform a wifi scan.
Only when asked to or when to perform a reconnect from an "unstable" connection or when the last used connection is not available any more.

So even if this will be implemented, then it will hardly ever trigger such an event.
This means you cannot use it for IFTTT kind of rules. (or you must perform scans at a regular interval)

blb4github, 2019-12-22:
It would be fine to perform scan as part of a timer rule. Something like:
On Rules#Timer=1 do
 if wifiscan#<SSIDtoCheck>=1
  Publish <topic>,[wifiscan#<SSIDtoCheck]
 endif
 timerSet,1,300
endon

TD-er, 2019-12-23:
A scan does cause connection issues and probably a reconnect.
What can be done is spit the scan into a scan per channel, which only takes roughly 200 msec.

blb4github, 2019-12-24:
Per channel would be sufficient, the networks I want to detect do have fixed channel.

TD-er, 2019-12-25:
OK, then it may be a useful feature I guess.
So in short, what you want is:
- Command to initiate a scan with optional the channel number
- Scan will be async, sending out an event per found SSID
- Event must have some parameters like SSID, channel, RSSI.

Right now, we do not have proper support yet for strings as an event parameter, so that's something that has to be implemented first.

blb4github, 2019-12-25:
Sounds great to me. About the strings as event parameter: Maybe it helps that in my use case I know which SSID I'm searching for. Maybe it's easier to initiate a scan with the SSID I'm interested in as a parameter and get 0 back when SSID not in reach and 1 or the RSSI value when in reach.