letscontrolit/ESPEasy #2833: Next stable release?

SANCLA, 2019-12-24:
Dear developers,

You recently starting asking for funding to focus more on developing this great peace of software.
And so I did, I have just chosen to be a patreon for this project.

However, it does raise a question with me.
The last stable release is version R120, august 2016 (more then 3 years ago!).
The next in line would be a release candidate version v.2.0 from 2018 (more then 1,5 years ago!)

When can we expect a new stable release, one that is getting the support it needs (bugfixes)?
Why work even with version numbers if you don't release?

Currently, it seems this project has been diminished to a permanent Alpha state, or so it seems.

Can we, now that this project is being funded, expect stable releases and a roadmap?
Or am I asking a question that should not be asked for this project?

Grovkillen, 2019-12-24:
Thank you for being a Patreon, we welcome you work open arms.

And thank you for raising this question, I'll try to answer it here.

About 2.5 years ago I got involved in this project and started adding info to the wiki. About that time Gijs (TD-er) also got involved. Around that time the initial developer had let the project go since he didn't have the commitment to take it further. So a third person took over but only lasted for 6 month (give or take). Me and Gijs decided to take over at that point. Long story short: we ended up redrawing the goal/vision for this project since we wanted this to be more than just a firmware.

The vision is something like this:
* Break away the backend (low level code) from the frontend
* A unit shouldn't need the frontend to function
* Only one unit in a group of units need to have a frontend, browse to that unit and then use that frontend to configure all the other units
* Refactoring of the source code to make it more modular
* Make the wifi implementation stable
* Make the unit asynchronous, no blocking code
* Make plugins less independent in terms of communication (you shouldn't need to write com-specific code) = abstract layers
* Integrate wiki into the frontend = get the help where you need it
* Add dashboards for the frontend
* Make the frontend modular
* Make this into a serious venture, try to make it our daytime job

Of these points we have come a long way to fulfill but to do so we had to really completely understand the code base. Gijs has been adding great input to the core team of Espressif in terms of stability (this means that all other projects that are relying on the core has been getting help from us). And to speak from my side only the last week's or so we've been feeling that we finally has tamed this beast.

We've been discussing the release and even if I cannot give you an exact date today I can tell you that it's not too much time left. 

And in terms of versioning, I can only agree with you. We where given this project with the version 2.0 and since that one was never released but unofficially used we decided to start fresh and release our first version under the version of 2.1.

@TD-er : do you have anything to add? Perhaps we could start a little list of things we will need to have done before the 2.1 release?

2.2(?) will include the new GUI ...

SANCLA, 2019-12-24:
How can I help?

Strong points are mainly organization, documentation and translation Dutch/English.
I'm more a junior at development and testing.

SANCLA, 2019-12-24:
Btw, thanks for that enlightening answer, I really appreciate you took the time to explain that bit of history and the current way forward!

TD-er, 2019-12-25:
My addition to the points Jimmy made.
I want to stress that although we want to split the fronted/backend, we surely want to keep the possibility to still use a single node with fronted/backend.

One of the things I really regret is that we took on too many changes at once which made it very hard to find out what was causing the instability issues.
So we tackled them one at a time.

Right now I have got the feeling things are back to stable again and that's also why we now do less builds at a time, have test builds etc.
Just to make sure that if issues arise we know where they came from.

I was hoping to put the "release tag" on it by Christmas (today), but since I am in the middle of moving to a new house, so let's hope we don't find any alarming issues in the next days and maybe we can call it a 2.0 in 2020-01-01 :)

tonhuisman, 2023-09-02:
This is not really relevant anymore, so can be closed.

closed 2023-09-02 by TD-er