letscontrolit/ESPEasy #2836: [json] endpoint "buildinfo" declares plugins twice

Grovkillen, 2019-12-29:
```
{
  "plugins": [
    {
      "id": "1",
      "name": "Switch input - Switch"
    },
    {
      "id": "2",
      "name": "Analog input - internal"
    },
    {
      "id": "4",
      "name": "Environment - DS18b20"
    },
    {
      "id": "26",
      "name": "Generic - System Info"
    },
    {
      "id": "28",
      "name": "Environment - BMx280"
    },
    {
      "id": "33",
      "name": "Generic - Dummy Device"
    },
    {
      "id": "36",
      "name": "Display - OLED SSD1306/SH1106 Framed"
    },
    {
      "id": "37",
      "name": "Generic - MQTT Import"
    },
    {
      "id": "49",
      "name": "Gases - CO2 MH-Z19"
    },
    {
      "id": "52",
      "name": "Gases - CO2 Senseair"
    },
    {
      "id": "56",
      "name": "Dust - SDS011/018/198"
    },
    {
      "id": "59",
      "name": "Switch Input - Rotary Encoder"
    },
    {
      "id": "82",
      "name": "Position - GPS [TESTING]"
    },
    {
      "id": "85",
      "name": "Energy - AccuEnergy AcuDC24x [TESTING]"
    },
    {
      "id": "87",
      "name": "Communication - Serial Proxy [TESTING]"
    }
  ],
  "controllers": [
    {
      "id": "1",
      "name": "Domoticz HTTP"
    },
    {
      "id": "2",
      "name": "Domoticz MQTT"
    },
    {
      "id": "3",
      "name": "Nodo Telnet"
    },
    {
      "id": "4",
      "name": "ThingSpeak"
    },
    {
      "id": "5",
      "name": "Home Assistant (openHAB) MQTT"
    },
    {
      "id": "6",
      "name": "PiDome MQTT"
    },
    {
      "id": "7",
      "name": "Emoncms"
    },
    {
      "id": "8",
      "name": "Generic HTTP"
    },
    {
      "id": "9",
      "name": "FHEM HTTP"
    },
    {
      "id": "10",
      "name": "Generic UDP"
    },
    {
      "id": "11",
      "name": "Generic HTTP Advanced [TESTING]"
    },
    {
      "id": "12",
      "name": "Blynk HTTP [TESTING]"
    },
    {
      "id": "13",
      "name": "ESPEasy P2P Networking"
    },
    {
      "id": "14",
      "name": "Homie MQTT (Version 4.0.0 dev)"
    },
    {
      "id": "15",
      "name": "Cache Controller [Experimental]"
    },
    {
      "id": "16",
      "name": "Zabbix"
    },
    {
      "id": "17",
      "name": "LoRa TTN - RN2483/RN2903 [TESTING]"
    }
  ],
  "notifications": [
    
  ],
  "build": "20104",
  "git-build": "",
  "system-libraries": "ESP82xx Core bc204a9b, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support",
  "plugins": "15",
  "plugin-description": " [Normal]"
}
```
The last plugin will override the other plugin path so only the value "15" is left... remove the last plugin path and it will be all right.

TD-er, 2019-12-29:
Or change to "nr-plugins" ?

Grovkillen, 2019-12-29:
I get the number of plugins for free since it's an array but sure :)

TD-er, 2019-12-29:
While we're at it, the "plugin-description" should be trimmed.

closed 2020-02-21 by TD-er