letscontrolit/ESPEasy #2837: I found 3 bugs :(

Jufo84, 2019-12-29:
Hardware:
Wemos D1 Mini (esp8266, 4MB flash).

Software:
ESP_Easy_mega, normal_ESP8266 for 4MB flash (4M or 4M1M.bin), tested also newest version mega-20191208.

Bugs:
1. ESPEasy P2P Networking, does not send value to other devices, last working version: mega-20190805;
2. ESPEasy P2P Networking sometimes crashing when sending values to LCD (no refresh data on LCD). I dont saw problem in old system with old GlobalSync. Restarting device resolve this problem temporarily to next crash;
3. LCD not display values when sending command over http (cmd, LCD), latest working version mega-20191123). Probably it problem with some string.

Best Reagards,
Robert

TD-er, 2019-12-29:
Point 1:
Have you tried these steps as described in the [documentation](https://espeasy.readthedocs.io/en/latest/Controller/C013.html#sharing-plugins-among-nodes) ?

Point 2:
What exact plugin settings?
What is crashing? the node or no longer sending p2p values? Is the unit number unique? Always the same IP address? 
This sounds like a separate bug, in need of its own issue.

Point 3: 
Have you checked this issue: https://github.com/letscontrolit/ESPEasy/issues/2724 ?

Extra point:
Please list issues in their own issue as it may be very hard to handle "mixed issues" like this

Jufo84, 2019-12-30:
Point 1: yes

Point 2.
It's very simple, temperature value from virtual sensor (ESPEasy P2P Networking) is display on LCD (i2c, 2x16);
Sometimes problem is with display new values on LCD, show only old value, but when I restart device its works normally to next freeze (crashing).
Yes, units numbers is unique. I use static IP addresses on devices the same as dhcp leases.
But I can't provide more information right now because the device is installed in a family home in another city.

Point 3.
I send values from domoticz to esp8266, for example (part of script):
`def get_domoticz_sensor(idx):
    url = "http://" + str(cfgAuth) + str(cfgDomoticzHost) + "/json.htm?type=devices&rid=" + str(idx)
    response = urllib.urlopen(url);
    data = json.loads(response.read())
    if data["status"] == "OK":
        return data["result"][0]

#Create screens
def ESP_KITCHEN_1():
    c.setopt(c.URL, str(cfgEspKitchen) + "lcdcmd,clear")
    c.perform()
    c.setopt(c.URL, str(cfgEspKitchen) + "lcd,1,1,TEMP1%20OUT:%20" + str(get_domoticz_sensor(29)["Temp"]) + str(cfgDeg))
    c.perform()
`

TD-er, 2019-12-30:
When the node stops updating values, via p2p, can you check:
- both nodes are reachable via web page?
- Wifi has just reconnected?
- Does it start sending out p2p packets again when you ping the unit?

What do you mean by "static IP" ?
- The unit gets assigned the same IP via DHCP?
- It is set to a fixed IP and thus no DHCP is used on the node?

closed 2022-01-08 by RobertWojtowicz