letscontrolit/ESPEasy #2838: Weather information from WeatherStack as a plugin

Grovkillen, 2019-12-31:
https://weatherstack.com/documentation

Weatherstack has given us a big thumbs up to use their API. The free plan allow for 1000 requests per month (we should monitor this) and you can fetch it using HTTP.

API Token is 32 characters long

Grovkillen, 2019-12-31:
**Scope of plugin**

current endpoint:
* temperature
* weather_code
* wind_speed
* wind_degree
* pressure
* precip
* humidity
* cloudcover
* feelslike
* uv_index
* visibility

I have asked them to provide a timestamp in epoch (now it's a AM/PM string) as well.

Grovkillen, 2019-12-31:
weather should be fetched using the internal lat/long or plugin set lat/long

TD-er, 2019-12-31:
Free plan per user I assume?

It looks like you can convert the units your receive by adding the requested unit of measure in the API call.
So the selector should also include that.

Maybe we can include this to use an async http call, like the planned async sendtohttp we plan to have in the rules.

Grovkillen, 2019-12-31:
Yep it's per user.

Grovkillen, 2019-12-31:
Historical/Forecast is paid plan so perhaps that can be added as a selector as well?