letscontrolit/ESPEasy #2839: MCP23017: Internal Pull-up Resistors not re-enabled on reboot.

Humanoidx, 2020-01-02:
### Summarize of the problem/feature request
The 16 port GPIO Expander: MCP23017 has internal pull up resistors that are enabled by default when initially setting up the devices. However, upon either esp8266 or mcp23017 reboot... the pull-up resistors are not re-enabled and revert to unstable floating inputs.

### Expected behavior
The internal pull-up resistors of the MCP23017 should be re-enabled on either esp8266 or mcp23017 reboot
### Actual behavior
The internal pull-up resistors of the MCP23017 are not re-enabled upon reboot

### Steps to reproduce
1. Connect MCP23017 to ESP8266
2. Add Device Switch Input: MCP23017 
3. You will notice that the internal pull-up resistors work as expected
4. Perform soft or hard reboot and you will notice the Internal pull-up resistors no longer work

### System configuration
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core bc204a9b, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20191208
Plugins:⋄
46 [Normal]
Build Time:⋄
Dec  8 2019 18:04:38
Binary Filename:⋄
ESP_Easy_mega-20191208_normal_ESP8266_4M1M.bin

Hardware: NodeMCU

ESP Easy version:

giig1967g, 2020-01-02:
Hi,
can you explain how you check if pull-up is enabled or not?

In terms of coding, the code executed during the initial setup is the same code executed after the reboot. So I can't see how it could have different behavior.

Maybe the unit is defective?

Humanoidx, 2020-01-02:
Ive checked the pullup resistors by using a test lead... connecting one end
to ground and the other end took the MCPs GPIO pins.

When internal pullups are enabled,  the default gpio value is 1. When
connected to ground,  the value drops to 0. When ground is removed,  the
value immediately reverts to 1.

Upon restart, the values are mostly 0, when connected to 3.3v ... the value
changes to 1. Upon removal of 3.3v, the value doesn't immediately change,
but takes some time  or a physical flick usually does trick.

If I go back into devices when it is acting like there is no pull up
resistors, I can simply select "Submit" and the value changes to 1 and acts
stable as if it's internal pullup resistors have been re-enabled. I must do
this individually for each gpio.

Thank you for your quick response.

uzi18, 2020-01-02:
It's looks like not best test for it.

Humanoidx, 2020-01-03:
Can you recommend a better test?
When the devices are created the gpio act if they are pulled up.
upon restart they act as if they are floating
upon resubmitting the device (with no changes) they act as if they are
pulled up again.
Seems like software issue rather than hardware.
You can configure the MCP's pullup resistors via the i2c bus, my assumption
was that espeasy wasnt reconfiguring this after a reboot and only when the
submit button was pressed.

Humanoidx, 2020-01-19:
Turns out this is the behavior when not setting the MCP reset pin to hi..... ooops

closed 2020-01-19 by Humanoidx