letscontrolit/ESPEasy #2842: Pulse Counter only available at task 1 to 3, want to use more.

TBobsin, 2020-01-04:
### I have...
- [ yes] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ yes] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [yes ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ yes] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ yes] Tested previous/other build (mention which one already tested) seems to be an old bug.
 look at this https://forum.fhem.de/index.php/topic,66513.msg727263.html#msg727263
- [x ] Tested on other node to make sure hardware isn't defective. Works with an switch Configuration
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request
Pulse Counter only available at task 1 to 3, want to use more.
There is no Information on the device that only task 1 to 3 is possible to use.
Pulse Counter should be available on all IRQ able GPIOs.

Task 4 to 12 can be configured to Generic Pulse Counter, but the Counter dos nothing. an parallel configured Switch shows the (of course slow) correct input level. So Hardware is functional. 

Firmware
Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191208
Plugins:⋄ | 79 [Normal] [Testing]
Build Md5: | 1fa38dd022aa10d76393c8108f922a
Md5 check: | passed.
Build Time:⋄ | Dec  8 2019 18:17:48
Binary Filename:⋄ | ESP_Easy_mega-20191208_test_ESP8266_4M1M_VCC.bin

### Expected behavior

Pulse Counter should work on all task if the gpio supports the irq

### Actual behavior
You could configure the counter, but the counter dos not count.

### Steps to reproduce

1. configure a counter at task 4 to 12 

Does the problem persists after powering off and on? yes

### System configuration

Hardware: NODE MCU 

ESP Easy version: Git Build:⋄ | mega-20191208

### Rules or log data
rules not activated, nothing configured here.

TD-er, 2020-01-04:
The pulse counter only registers the interrupt calls for the first few tasks.
So it is a known issue, but I guess it is a valid request to increase the number of tasks for this.

TD-er, 2021-06-05:
It is fixed in the most recent update of the pulse plugin.
So I will close it.

closed 2021-06-05 by TD-er