letscontrolit/ESPEasy #2843: Multiple levels of access control

TD-er, 2020-01-05:
Right now we only have the "admin" user which can have a password set.
The main criterium here is to have the web interface protected where you can change settings.

However, it would be useful to have a bit more access level control, like "view" and "command".
Also we still have the bug, where flashing of the firmware is still not password protected.

Related issues:
- https://github.com/letscontrolit/ESPEasy/issues/2535
- https://github.com/letscontrolit/ESPEasy/issues/118
- Forum: [HTTP control without password?](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=6702)

So in short:
- Need more levels of security (view/admin)
- Check if subnet block can be enhanced (e.g. what to do with VPN connections)
- Add pass option for http commands
- Check if other access entries for commands can be protected (MQTT, p2p, serial, etc)

ili, 2021-12-06:
[HTTP control should be authorized](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=8880&p=56255#p56157), even using Basic auth not only for UI but for `/control ` too would be great step.

ironataerial, 2021-12-07:
I am adding this comment to "push attention up a little"

ili, 2021-12-10:
Smth like `http://<espeasyip>/control?cmd=<command>&p=<secret key>` would be nice too (even easier to configure for integrators)

TD-er, 2021-12-10:
Password or secret in a GET URL over non-encrypted http might not be the best idea.
But it is more than what we have now, so we can consider it...

ili, 2022-03-04:
> Password or secret in a GET URL over non-encrypted http might not be the best idea. But it is more than what we have now, so we can consider it...

Certainly it is not quite secure, but it is much better then nothing :)