letscontrolit/ESPEasy #2847: different behavior in Analong input in "normal" firmware an in "test" firmware

flucasro, 2020-01-07:
I am using ESP_Easy_mega-20191208_normal_ESP8266_4M1M.bin
and ESP_Easy_mega-20191208_test_ESP8266_4M1M_VCC_MDNS_SD.bin

And Analog input on an ESP8266 (wemos mini)

The test package returns values of
882 ≙ 66418.727
1074 ≙ 66610.203 (over 1023!)
2015 ≙ 67551.000

Enabling/Disabling oversampling does not have any effect.
These values are obviously wrong!

And the normal package returns values such as
533 ≙ 533.000
Within the expected range of range 0-1023.

TD-er, 2020-01-07:
Looks like there is something wrong with the settings used.
It looks like the plugin in the "test package" is actually using some kind of calibration values.
Can you make sure to save again and see if that helps?

flucasro, 2020-01-07:
[image]

I cannot see any changes using different calibration values, or enabling/disabling it

flucasro, 2020-01-07:
[image]

TD-er, 2020-01-07:
Just checking, but what hardware is this?

flucasro, 2020-01-07:
WEMOS MINI 4 MB

TD-er, 2020-01-07:
Are you sure you are not running a "_VCC" build?

flucasro, 2020-01-08:
I was using 
 am using ESP_Easy_mega-20191208_normal_ESP8266_4M1M.bin
and ESP_Easy_mega-20191208_test_ESP8266_4M1M_VCC_MDNS_SD.bin

if Using ESP_Easy_mega-20191208_test_beta_ESP8266_4M1M.bin
There are no problems

tonhuisman, 2023-09-02:
The issue seems no longer relevant, as the wrong build was used (VCC). Can be closed.

closed 2023-09-02 by TD-er