letscontrolit/ESPEasy #2848: Command Error Handling

psi-4ward, 2020-01-07:
There should be way to do some error-handling in Rules:

Example: `SendTo,2,"LongPulse,14,1,10"` in a P2P Network
* OnSuccess: Flash the green LED
* OnFailure (ie Unit 2 not available): Flash the red LED

Example: `SendToHTTP,url.com,80,/report.php?hash=123abc456&t=[temp2#out]`
* Error when url.com not available or request returns some bad response-code (>= 400) 

[Thread](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=7348)

Cause I'm very new to ESPEasy I suppress implemention-ideas.

TD-er, 2020-01-07:
Right now, only controllers have some error handling and commands indeed have none.
Related issue: https://github.com/letscontrolit/ESPEasy/issues/2796

psi-4ward, 2020-04-01:
So I managed some _ACK'ing_ by using two dummy devices (one each device).

Device 1 (Sender): 
```
on Taster#State=1 do
  // Update dummy_reciver and trigger TaskRun
  SendTo,2,"TaskValueSet 12,1,1"
  SendTo,2,"TaskRun 12"
  // orange Neopixel while waiting for ACK
  NeoPixel,1,200,133,0
  // Reciver should set sender_dummy ack value within 1s
  timerSet,2,1
endon

// ACK handler
on Rules#Timer=2 do
  if [sender_dummy#ack]=1
    NeoPixel,1,0,200,0 // green
  else
    NeoPixel,1,200,0,0 // red
  endif
  timerSet,3,1
endon

// ACK final handler
on Rules#Timer=3 do
  TaskValueSet 12,3,0 // reset sender_dummy#ack
  NeoPixel,1,0,0,0
endon
```

Device 2 (receiver)
```
On receiver_dummy#run=1 do
  // reset run state
  TaskValueSet 12,1,0
  // send ACK to sender
  SendTo,1,"TaskValueSet 12,3,1"
endon
```

Yes not very slick but works (for my case) and visualizes if _receiver_-Device is unreachable.

uzi18, 2020-04-01:
You can use combo for taskvalueset and run in one command.
For such use you can use local variables they are more readable than dummys.

uzi18, 2020-04-01:
It is (theoreticaly) possible to check state on http request but on udp like in sendto could be problem without special case.

psi-4ward, 2020-04-01:
Do you have small examples for both cases?

uzi18, 2020-04-01:
https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#special-task-names <- use INT#/VAR# plus LET command

examples of use are below

command **taskvaluesetandrun** = taskvalueset +taskrun arguments are same as for taskvalueset
ach see no docs for this one in command reference