letscontrolit/ESPEasy #2849: Longpulse dont work

Basti77, 2020-01-09:
Hello,
I have noticed that there is again, or always a problem with longpulse.

I send a Longpulse command over fhem. it is recognized but not processed.

ESP Log:
209383: HTTP: longpulse,12,1,30
209388: SW : GPIO 12 Pulse set for 30 sec
211928: WD : Uptime 4 ConnectFailures 0 FreeMem 20152

after mach tried to switch with longpulse the normal GPIO command doesn't work anymore. you have to restart esp. 

I Use a Sonoff S26 Plug.

different BIN files were tried (Bulids) 

last 
ESP_Easy_mega-20191208_normal_ESP8266_1M.bin

Does anyone have any idea what it could be?

TD-er, 2020-01-09:
Does sending long pulse work when you send it via the tools page?

Basti77, 2020-01-10:
Here is the output when I execute the ESP command.

Command Output

{
"log": "GPIO 12 Pulse set for 30 ",
"plugin": 1,
"pin": 12,
"mode": "output",
"state": 1
}
Log:
71770031: SW : GPIO 12 Pulse set for 30 sec

the command is accepted but not passed to the GPIO

the relay has not closed

TD-er, 2020-01-10:
@giig1967g Do you have any idea here?

giig1967g, 2020-01-14:
@TD-er 
will look at it

giig1967g, 2020-01-16:
@Basti77 
I tested with version 20191208 on a Wemos board and the command works fine for me.
Can you try if the GPIO command works fine or not?
Maybe a wiring issue?

Basti77, 2020-01-16:
Via the console GPIO 12.1 or 12.0 works 
Via the console Longpulse 12,1,10 
is accepted in the log but nothing happens.
after that you can't switch over the normal GPIO command. then only a reboot helps.

on an old 2018 build and a Sonoff Basic it works by default, but not on the Sonoff S26.

TD-er, 2020-01-16:
Have you tried the test build I linked here in this PR: https://github.com/letscontrolit/ESPEasy/pull/2778#issuecomment-567685164 ???

Basti77, 2020-01-22:
Sorry, I took a different path after all that trying. 
I installed Tasmota and then I solved the problem.

Grovkillen, 2020-01-22:
That's no good for us :(

giig1967g, 2020-01-22:
The OP said that it was working on Sonoff Basic, but not on S26.
Unfortunately I don't have a Sonoff S26 to try and replicate the problem.

duesed4, 2020-04-18:
I also have problems with longpulses on NodeMCU. I use a selfcompiled Version from April 14th from the repository mit mDNS activated (does also not work propperly, but should not be the Problem).
I use ESPEasy with FHEM. After Sending longpulse no reaction in FHEM is visible. I think, I heard my relay clicking some times. After trying a longpulse also on/off seems to have problems und states are not always send to fhem or at least delayed.

uzi18, 2020-04-19:
@duesed4 you should check your controller configuration, please attach screenshot

tonhuisman, 2023-09-02:
This seems outdated by the refactoring of all GPIO and LongPulse related code. This can be closed.

closed 2023-09-02 by TD-er