letscontrolit/ESPEasy #2850: Wifi Disconnects - Sonoff Basic

Basti77, 2020-01-10:
During troubleshooting for Logpulse, the last version installed on a Sonoff Basic.

ESP_Easy_mega-20191208_normal_ESP8266_1M.bin

I immediately noticed that the Wifdi connection is extremely unstable with this image.
I downgrade to version 2.0X and the connection is good again.

Is this a known problem?

Grovkillen, 2020-01-10:
Did you do a upgrade on that unit or was it a clean install?

Basti77, 2020-01-10:
I flashed the blank,1024 bin file previously.

Grovkillen, 2020-01-10:
Ok, thanks.

TD-er, 2020-01-10:
Could you test a build from around end of September?

Basti77, 2020-01-11:
I just tried the build ESPEasy_mega-20190928.
no improvement, but I think it is not the wifi connection.
According to the log of my Aruba WiFi infrastructure the connection is stable, there must be problems on IP level.
(On a normal plastic router I have also tested it, the same behavior.)
Can the internal firewall in ESP be the problem? 
(in the ESPEASY version 2018*** there was no filtering on network level) 
Different chip in the Sonoff Basic?
The board says: Sonoff TH_V1.1

the S26 has no problems with interruptions in any build.

TD-er, 2020-01-11:
The filtering should not be a problem, unless you will be switching subnets (e.g. using a vpn or using an access point as a router and not a bridge to your network)
That filtering only applies to the website access, it should still work fine with controllers etc.

Basti77, 2020-01-11:
> That filtering only applies to the website access, it should still work fine with controllers etc.

I can't confirm that.
it filters all IP traffic. A ping can't get through from another subnet.
My IoT devices have their own Wifi and network.

But for testing I was also in the network with my notebook. Everywhere the same problem.

TD-er, 2020-01-11:
Then it isn't the filter I added roughly 1.5 years ago (or was it 2 years already).
That subnet filter is only acting on the web pages.

If you even can't ping, then we have a different issue at hand here.
It sounds like the node doesn't receive an IP address. (or doesn't set it active)

Basti77, 2020-01-11:
I've tried DHCP and static IP. It makes no difference.

Sporadically, a ping comes through.
If the subnet filtering (default) is on you can never ping from the normal client net. therefore my question for an internal fw rule.

i found this old thread.
https://github.com/letscontrolit/ESPEasy/issues/2088

the plastic router test only had letters as wpa key.

In the productive IoT wifi i use # and ! as special characters.

tonhuisman, 2023-09-02:
This is most likely outdated by many improvements in the WiFi department. Issue can be closed.

closed 2023-09-02 by TD-er