letscontrolit/ESPEasy #2851: [Patreon] we need to a place to store secret keys for the upcoming "sponsors only" feature set

Grovkillen, 2020-01-10:
I suggest security.dat

Patreon ID: **int32** (8 digits long currently)
UUID: **int64**?

UUID is generated by our server and is unique to each Patreon.

TD-er, 2020-01-10:
Usually an UUID is 128 bit, which may be a bit overkill, but that's how long UUID's are.

Grovkillen, 2020-01-11:
You're the boss :)

Misiu, 2020-01-15:
@Grovkillen what will be those features? Maybe listing them will help you guys reach the goal quicker 🙂

Grovkillen, 2020-01-15:
@Misiu Good idea!

This list is preliminary and might not apply to all tiers on Patreon. Some will be for bronze, some to silver level tiers. Gold and above is probably the level that will not add extra features to the OS but more of "behind the scenes" and things like that.

In the GUI
* Theme your interface
* Create dashboards

On the future site (cloud)
* Compile special builds
* Backup nodes
* Mass deploy (settings, updates)
* Automatic updates

Misiu, 2020-01-15:
@Grovkillen thanks for the list!
Compiling builds on your site and automatic updates/deploy would be a fantastic feature. I think not only I will consider changing my tier on Patreon.

henkiejan1, 2020-01-18:
The last  beta release is from 8 december. Is this the reason maybe there is no new version recently? :)

Grovkillen, 2020-01-18:
@TD-er have moved to a new house... That's the main reason. I'm still developing the GUI.

TD-er, 2020-01-18:
Yep only yesterday I finally got to the point where I could setup some very basic breadboards on my desk again and it was also the first time in > 3 weeks I did program a single line of code.
(well program is a big word for commenting lines debugging the static IP => crash bug)